e2nIEE/pandapower #1118: Issues with pf_res_plotly when trying to plot on a Map

alekkk94, 2021-01-25:
Hello,

I am having an issue in plotting the results with pf_res_plotly with the on_map boolean being set to "True". On the other side, the simple_plotly function works perfectly and plots my network on the map. Moreover, if i try pf_res_plotly without a map, it works.
Are you aware of such a bug or this is something related to my machine?

Thank you,
Aleksandar

jwiemer112, 2021-01-28:
Hey, 

thanks for your post. There is also another issue (#1038) with Latitude and Longitude getting inverted in the plotly plotting functions. I made a **pull request (#1122 )** and tried to solve this issue. Feel free to check it out and giving me feedback on it.

Normally` pf_res_plotly` and `simple_plotly` should output the same network topology if on_map is True/False. Are there differenes when calling the functions in your script? Do you use a projection or are your coordinates in WGS84-Format? (epsg:4326).

I plotted a network with WGS84 Coordinates using both functions. The position and coordinates were at both cases inverted.

At the moment using a projection mixes the lat/lon values (Bug). However the draw_traces function mixes them up also. That means if you use a projection the output should be correct, because (-1)*(-1) = 1, using no projection only inverts the coordinates 1 time (in draw_traces) so the grid will be plotted 90° off. 

If you dont use any projection the output from simple_plotly and pf_res_plotly should be the same in terms of grid topology and the position of the grid on the map. It should be inverted once (Bug).

Its a bit confusing I hope you understand everything. 

Greetz Jan

rbolgaryn, 2021-02-23:
Hi @alekkk94 ,

can you please try with the latest pandapower from develop branch and let us know if the issue is still relevant?

alekkk94, 2021-02-25:
Hi @rbolgaryn and @jwiemer112 jwiemer112, thank you for your responses - i am using a specific projection for the observed region in meters. Basically, when comparing simple_plotly with map projection and pf_res_plotly without a map projection -> the map is inverted/rotated. So, I think you're right regarding the issue with the mixture of coordinates. However, it's strange that when i try to use pf_res_plotly with a map projection, there is nothing ( no map/nodes/lines). @rbolgaryn to answer your question, i tried downloading the latest develop branch and it still doesn't work. @jwiemer112, was your fix implemented in the develop branch or should I manually insert it? 

Thank you again !

jwiemer112, 2021-02-25:
Hey @alekkk94,  thanks for your answer :). My changes are already integrated in the latest pandapower(develop) commit. 

Do you see the network plot if you zoom out? 

Here is an example with the oberrhein network from pandapower. It uses coordinates in the Gauß-Krüger projection. 

```
import pandapower as pp
from pandapower.networks import mv_oberrhein
from pandapower.plotting import simple_plot, simple_plotly, pf_res_plotly
from pandapower.plotting.plotly.mapbox_plot import set_mapbox_token
set_mapbox_token('')

net = mv_oberrhein()
simple_plotly(net, on_map=True, map_style='streets', projection='epsg:31467')
pf_res_plotly(net, on_map=True)
```

This code works for me. However you have to insert your own token. Keep in mind that you have to erase the second projection of pf_res_plotly() if you use both methods directly after each other. Thats because the first projection alrdy transforms the coordinates to WGS84. If you call it then a second the transformation messes its up.

Here are the outcoming files from the plots, the network takes place in south germany, near to the city "Freiburg".
[image]
[image]

jwiemer112, 2021-02-25:
To update the pandapower repository, open the git bash terminal, move to the pandapower folder. Make sure in brackets the develop branch shows up -> pandapower(develop). Then type `git pull origin develop`. type `gitk` in the git bash after that to see the history of all commits. My pull request was merged at the 17.02.2021. The hash is: 528d09fe347bab7ed69cb90d89551016a2970f01

jwiemer112, 2021-02-25:
Checkout the Installation guide on our Homepage if you have problems to set up pandapower with git.

https://www.pandapower.org/start/

Choose **Development Version** on the left menu.

Greetz Jan

alekkk94, 2021-02-25:
@jwiemer112 Thank you very much, that was actually the issue - I was inserting the projection in both, simple_plotly and pf_res_plotly, which was effectively messing up the second one. Now the issue is fixed and it works well!

jwiemer112, 2021-02-25:
Yeah! Feel free to close this issue.

Greetz Jan

closed 2021-03-04 by rbolgaryn