e2nIEE/pandapower #1124: Excess voltage magnitude when no load

edxu96, 2021-02-02:
Thanks for wonderful package.

I am calculating AC power flow for some feeders with 60, 10 and 0.4 kV voltage levels at the same time. Voltage magnitudes (VMs) are my primary concern. This is the result for feeder 406, compared with that from power factory (buses are sorted according to power factory result):

[image]

__I am not sure if the result from power factory is correct.__ So I tried power flow when load is not present, and VMs are above 1.0 p.u. except the slack bus.

[image]

ascheidl, 2021-02-02:
hi,

you should check that the transformer vn_hv_kv, vn_lv_kv, and tap_* values are equal to powerfactory.

also probabily the busses are not sorted in the same order in your plot ?

edxu96, 2021-02-02:
@ascheidl Thanks for the reply. I just triple-checked using `print(net.trafo)`. `vn_hv_kv` and `vn_lv_kv` are correct. There is no tap changer. Also, I used the same set of parameters in `PyPSA` and it was okay.

> also probabily the busses are not sorted in the same order in your plot ?

The main problem I think is shown by the second figure. Even when there is no load, VMs are not 1.0 p.u.

ascheidl, 2021-02-02:
can you:

print(net.trafo)
print(net.bus.loc[net.trafo.hv_bus])
print(net.bus.loc[net.trafo.lv_bus])

and post here ?

edxu96, 2021-02-02:
I transposed `net.trafo`:

```
                          0          1          2          3          4          5   ...         20         21         22         23         24       25
name               N1_N21452  N1_N20595  N1_N20467  N1_N21429  N1_N21075  N1_N21235  ...  N1_N20465  N1_N21391  N1_N21241  N1_N20477  N1_N20426   N1_SVL
std_type              10-400     10-400     10-400     10-400     10-400     10-400  ...     10-630     10-630     10-630     10-630     10-630  60-8000
hv_bus                    23         26         11         18         21         33  ...          5         22         31         27         14        0
lv_bus                   198        249         43        115        127        252  ...         94         82         65        133         90       30
sn_mva                   0.4        0.4        0.4        0.4        0.4        0.4  ...       0.63       0.63       0.63       0.63       0.63      8.0
vn_hv_kv                10.0       10.0       10.0       10.0       10.0       10.0  ...       10.0       10.0       10.0       10.0       10.0     60.0
vn_lv_kv                 0.4        0.4        0.4        0.4        0.4        0.4  ...        0.4        0.4        0.4        0.4        0.4     10.0
vk_percent               4.0        4.0        4.0        4.0        4.0        4.0  ...        4.0        4.0        4.0        4.0        4.0     10.0
vkr_percent            1.075      1.075      1.075      1.075      1.075      1.075  ...   1.015873   1.015873   1.015873   1.015873   1.015873      0.6
pfe_kw                  1.15       1.15       1.15       1.15       1.15       1.15  ...        1.5        1.5        1.5        1.5        1.5     12.0
i0_percent           0.28751    0.28751    0.28751    0.28751    0.28751    0.28751  ...   0.238105   0.238105   0.238105   0.238105   0.238105  0.15001
shift_degree             0.0        0.0        0.0        0.0        0.0        0.0  ...        0.0        0.0        0.0        0.0        0.0      0.0
tap_side                 NaN        NaN        NaN        NaN        NaN        NaN  ...        NaN        NaN        NaN        NaN        NaN      NaN
tap_neutral              NaN        NaN        NaN        NaN        NaN        NaN  ...        NaN        NaN        NaN        NaN        NaN      NaN
tap_min                  NaN        NaN        NaN        NaN        NaN        NaN  ...        NaN        NaN        NaN        NaN        NaN      NaN
tap_max                  NaN        NaN        NaN        NaN        NaN        NaN  ...        NaN        NaN        NaN        NaN        NaN      NaN
tap_step_percent         NaN        NaN        NaN        NaN        NaN        NaN  ...        NaN        NaN        NaN        NaN        NaN      NaN
tap_step_degree          NaN        NaN        NaN        NaN        NaN        NaN  ...        NaN        NaN        NaN        NaN        NaN      NaN
tap_pos                  NaN        NaN        NaN        NaN        NaN        NaN  ...        NaN        NaN        NaN        NaN        NaN      NaN
tap_phase_shifter      False      False      False      False      False      False  ...      False      False      False      False      False    False
parallel                   1          1          1          1          1          1  ...          1          1          1          1          1        1
df                       1.0        1.0        1.0        1.0        1.0        1.0  ...        1.0        1.0        1.0        1.0        1.0      1.0
in_service              True       True       True       True       True       True  ...       True       True       True       True       True     True

                name  vn_kv type  zone  in_service
23  N1_N21452_layer1   10.0    b  None        True
26  N1_N20595_layer1   10.0    b  None        True
11  N1_N20467_layer1   10.0    b  None        True
18  N1_N21429_layer1   10.0    b  None        True
21  N1_N21075_layer1   10.0    b  None        True
33  N1_N21235_layer1   10.0    b  None        True
4   N1_N20887_layer1   10.0    b  None        True
34  N1_N20406_layer1   10.0    b  None        True
25  N1_N20515_layer1   10.0    b  None        True
15  N1_N21081_layer1   10.0    b  None        True
28  N1_N21745_layer1   10.0    b  None        True
2   N1_N20019_layer1   10.0    b  None        True
13  N1_N20239_layer1   10.0    b  None        True
7   N1_N20157_layer1   10.0    b  None        True
36  N1_N20281_layer1   10.0    b  None        True
6   N1_N20476_layer1   10.0    b  None        True
9   N1_N21060_layer1   10.0    b  None        True
24  N1_N21074_layer1   10.0    b  None        True
35  N1_N21073_layer1   10.0    b  None        True
29  N1_N20126_layer1   10.0    b  None        True
5   N1_N20465_layer1   10.0    b  None        True
22  N1_N21391_layer1   10.0    b  None        True
31  N1_N21241_layer1   10.0    b  None        True
27  N1_N20477_layer1   10.0    b  None        True
14  N1_N20426_layer1   10.0    b  None        True
0      N1_SVL_layer0   60.0    b  None        True

                 name  vn_kv type  zone  in_service
198  N1_N21452_layer2    0.4    b  None        True
249  N1_N20595_layer2    0.4    b  None        True
43   N1_N20467_layer2    0.4    b  None        True
115  N1_N21429_layer2    0.4    b  None        True
127  N1_N21075_layer2    0.4    b  None        True
252  N1_N21235_layer2    0.4    b  None        True
100  N1_N20887_layer2    0.4    b  None        True
[... cut ...]
270  N1_N21060_layer2    0.4    b  None        True
189  N1_N21074_layer2    0.4    b  None        True
181  N1_N21073_layer2    0.4    b  None        True
188  N1_N20126_layer2    0.4    b  None        True
94   N1_N20465_layer2    0.4    b  None        True
82   N1_N21391_layer2    0.4    b  None        True
65   N1_N21241_layer2    0.4    b  None        True
133  N1_N20477_layer2    0.4    b  None        True
90   N1_N20426_layer2    0.4    b  None        True
30      N1_SVL_layer1   10.0    b  None        True
```

ascheidl, 2021-02-02:
looks OK ... how about:

t = net.trafo.query("vn_hv_kv > 30")
print(t)
print(net.bus.loc[t.hv_bus])
print(net.bus.loc[t.lv_bus])

edxu96, 2021-02-02:
I transposed `net.trafo`:

```
                        0        1        2        3        4        5        6   ...        19        20        21        22        23        24       25
std_type            10-400   10-400   10-400   10-400   10-400   10-400   10-400  ...    10-630    10-630    10-630    10-630    10-630    10-630  60-8000
hv_bus                  33       34        7       20        3       32        9  ...        36         6        19         5        15        10        0
lv_bus                 136      250       40      266      177      135      140  ...       168       173       201        80       229       179        1
sn_mva                 0.4      0.4      0.4      0.4      0.4      0.4      0.4  ...      0.63      0.63      0.63      0.63      0.63      0.63      8.0
vn_hv_kv              10.0     10.0     10.0     10.0     10.0     10.0     10.0  ...      10.0      10.0      10.0      10.0      10.0      10.0     60.0
vn_lv_kv               0.4      0.4      0.4      0.4      0.4      0.4      0.4  ...       0.4       0.4       0.4       0.4       0.4       0.4     10.0
vk_percent             4.0      4.0      4.0      4.0      4.0      4.0      4.0  ...       4.0       4.0       4.0       4.0       4.0       4.0     10.0
vkr_percent          1.075    1.075    1.075    1.075    1.075    1.075    1.075  ...  1.015873  1.015873  1.015873  1.015873  1.015873  1.015873      0.6
pfe_kw                1.15     1.15     1.15     1.15     1.15     1.15     1.15  ...       1.5       1.5       1.5       1.5       1.5       1.5     12.0
i0_percent         0.28751  0.28751  0.28751  0.28751  0.28751  0.28751  0.28751  ...  0.238105  0.238105  0.238105  0.238105  0.238105  0.238105  0.15001
shift_degree           0.0      0.0      0.0      0.0      0.0      0.0      0.0  ...       0.0       0.0       0.0       0.0       0.0       0.0      0.0
tap_side               NaN      NaN      NaN      NaN      NaN      NaN      NaN  ...       NaN       NaN       NaN       NaN       NaN       NaN      NaN
tap_neutral            NaN      NaN      NaN      NaN      NaN      NaN      NaN  ...       NaN       NaN       NaN       NaN       NaN       NaN      NaN
tap_min                NaN      NaN      NaN      NaN      NaN      NaN      NaN  ...       NaN       NaN       NaN       NaN       NaN       NaN      NaN
tap_max                NaN      NaN      NaN      NaN      NaN      NaN      NaN  ...       NaN       NaN       NaN       NaN       NaN       NaN      NaN
tap_step_percent       NaN      NaN      NaN      NaN      NaN      NaN      NaN  ...       NaN       NaN       NaN       NaN       NaN       NaN      NaN
tap_step_degree        NaN      NaN      NaN      NaN      NaN      NaN      NaN  ...       NaN       NaN       NaN       NaN       NaN       NaN      NaN
tap_pos                NaN      NaN      NaN      NaN      NaN      NaN      NaN  ...       NaN       NaN       NaN       NaN       NaN       NaN      NaN
tap_phase_shifter    False    False    False    False    False    False    False  ...     False     False     False     False     False     False    False
parallel                 1        1        1        1        1        1        1  ...         1         1         1         1         1         1        1
df                     1.0      1.0      1.0      1.0      1.0      1.0      1.0  ...       1.0       1.0       1.0       1.0       1.0       1.0      1.0
in_service            True     True     True     True     True     True     True  ...      True      True      True      True      True      True     True

    vn_kv type  zone  in_service
33   10.0    b  None        True
34   10.0    b  None        True
7    10.0    b  None        True
20   10.0    b  None        True
3    10.0    b  None        True
32   10.0    b  None        True
9    10.0    b  None        True
30   10.0    b  None        True
4    10.0    b  None        True
2    10.0    b  None        True
29   10.0    b  None        True
13   10.0    b  None        True
35   10.0    b  None        True
28   10.0    b  None        True
11   10.0    b  None        True
14   10.0    b  None        True
23   10.0    b  None        True
16   10.0    b  None        True
22   10.0    b  None        True
36   10.0    b  None        True
6    10.0    b  None        True
19   10.0    b  None        True
5    10.0    b  None        True
15   10.0    b  None        True
10   10.0    b  None        True
0    60.0    b  None        True

     vn_kv type  zone  in_service
136    0.4    b  None        True
250    0.4    b  None        True
40     0.4    b  None        True
266    0.4    b  None        True
177    0.4    b  None        True
135    0.4    b  None        True
140    0.4    b  None        True
160    0.4    b  None        True
202    0.4    b  None        True
64     0.4    b  None        True
145    0.4    b  None        True
68     0.4    b  None        True
191    0.4    b  None        True
256    0.4    b  None        True
226    0.4    b  None        True
91     0.4    b  None        True
130    0.4    b  None        True
190    0.4    b  None        True
222    0.4    b  None        True
168    0.4    b  None        True
173    0.4    b  None        True
201    0.4    b  None        True
80     0.4    b  None        True
229    0.4    b  None        True
179    0.4    b  None        True
1     10.0    b  None        True
```

---

I also checked the external grid: `p_mw = 0.047608` and `q_mvar = -0.271233`, which seem to be problematic. But in another feeder called 729 without load,  `p_mw = 0.035935` and `q_mvar = -0.033909`. VMs are 1. So probably the injection from the external grid is not causing the issue.

edxu96, 2021-02-02:
```
                        25
std_type           60-8000
hv_bus                   0
lv_bus                   2
sn_mva                 8.0
vn_hv_kv              60.0
vn_lv_kv              10.0
vk_percent            10.0
vkr_percent            0.6
pfe_kw                12.0
i0_percent         0.15001
shift_degree           0.0
tap_side               NaN
tap_neutral            NaN
tap_min                NaN
tap_max                NaN
tap_step_percent       NaN
tap_step_degree        NaN
tap_pos                NaN
tap_phase_shifter    False
parallel                 1
df                     1.0
in_service            True

   vn_kv type  zone  in_service
0   60.0    b  None        True

    vn_kv type  zone  in_service
33   10.0    b  None        True
```

ascheidl, 2021-02-02:
looks good, indeed

in fig2 there are busses that have 1.0 pu, can you identify them ?

edxu96, 2021-02-02:
There is only one bus with VM lower than 1.003 pu, and it is the 60 kV slack bus.

I added it using:

```
        pp.create_ext_grid(net=net,
                           bus=pp.get_element_index(net, "bus", SLACK[which]),
                           name='ExternalGrid_60kV',
                           va_degree=0,
                           in_service=True)
```

ascheidl, 2021-02-02:
and the 60/10 transformer is connected to this bus at the hv side (correct?) ...

for some reason the voltage rises over the transformer ... do you have larger infeed in the 10kV level ?

if so, maybe in powerfactory you calculate a different load case (different scaling factors)

edxu96, 2021-02-02:
Yes. Like `ExternalGrid_60kV  - slack_hv - 60/10 transformer - slack_lv - ...`

By the way, I forgot to mention. The 1.003 p.u. problem in fig2 disappears if I do `ExternalGrid_10kV - slack_lv - ...` by ignoring the 60/10 transformer:

[image]

> do you have larger infeed in the 10kV level ?

There is no other infeed in fig2.

> if so, maybe in powerfactory you calculate a different load case (different scaling factors)

Maybe we should ignore power factory results for now, because I am not sure if it is correct. The main issue, I think, is 1.003 p.u. when there is no load (and no generation) at all.

ascheidl, 2021-02-02:
what if you just use

extgrid - bus 0 - trafo 60/10 - bus 1

(only 2 busses + trafo + ext grid)

do you still geht this high voltage at the lv bus ?

edxu96, 2021-02-02:
No. 

```python
def test_transformer():
    net = pp.create_empty_network()

    pp.create_bus(net, name='bus0', vn_kv=60)
    pp.create_bus(net, name='bus1', vn_kv=10)

    hv_bus = pp.get_element_index(net, "bus", 'bus0')
    lv_bus = pp.get_element_index(net, "bus", 'bus1')
    pp.create_transformer(net,
                          name='trans',
                          hv_bus=hv_bus,
                          lv_bus=lv_bus,
                          std_type='60-8000')

    pp.create_ext_grid(net=net,
                       bus=hv_bus,
                       name='ExternalGrid_60kV',
                       va_degree=0,
                       in_service=True)

    pp.runpp(net)

    print(net.res_bus)
```

```
      vm_pu  va_degree   p_mw    q_mvar
0  1.000000   0.000000 -0.012 -0.000139
1  0.999995  -0.004286  0.000  0.000000
```

ascheidl, 2021-02-02:
is it possible for you to send me the grid model ? maybe that faster

edxu96, 2021-02-03:
Sorry for not replying in time. Here is a `pickle` file for the `pandapower` network. 

[feeder_406_pandapower.pkl.zip](https://github.com/e2nIEE/pandapower/files/5917740/feeder_406_pandapower.pkl.zip)

You can read it into your Python workspace directly, in case you don't know:

```
>>> import pickle
>>> with open('feeder_406_pandapower.pkl', 'rb') as input:
...     net = pickle.load(input)
...
>>> net
This pandapower network includes the following parameter tables:
   - bus (276 elements)
   - load (175 elements)
   - sgen (29 elements)
   - ext_grid (1 element)
   - line (249 elements)
   - trafo (26 elements)
```

ascheidl, 2021-02-03:
First, it doesnt look so bad. There is this 12 MW sgen which leads to a voltage rise in the grid:

[image]

However, without any load or sgen there is still some voltgae rise:

[image]

I checked your line properties and the c_nf_per_km seems quite high ... are the values correct ?

edxu96, 2021-02-04:
Many thanks. Will check those values.

The way you plot voltage magnitudes is really nice.

ascheidl, 2021-02-04:
it is part of pandapower :)

plotting.plot_voltage_profile(net)

edxu96, 2021-02-04:
> plotting.plot_voltage_profile(net)

I just tried it, but I couldn't find it in documentation.

ascheidl, 2021-02-04:
seems there is some documentation missing

[https://github.com/e2nIEE/pandapower/blob/develop/pandapower/plotting/powerflow_results.py](url)

import plotting
plotting.plot_voltage_profile(net)

should work

edxu96, 2021-02-04:
Yes, I have tried it in console. What I said is confusing.

It is not the documentation. Actually, there is doctring at all in file https://github.com/e2nIEE/pandapower/blob/master/pandapower/plotting/powerflow_results.py .

ascheidl, 2021-02-09:
is this solved @edxu96 ?

edxu96, 2021-02-09:
> is this solved @edxu96 ?

Thanks for asking. My colleague is working on it, re-calculating line parameters. Should be done this week.

Should we close this issue? Will reopen it if I found the problem remains.

edxu96, 2021-02-15:
Finally. We have checked parameters of cables and transformers.

I just found we confused "nano Farad" with "micro Farad" in an early excel table :-(

Thank you so much for being patient with this silly problem.

closed 2021-02-15 by edxu96