e2nIEE/pandapower #1130: Error when making PV to PQ conversion for some networks

DRL2019, 2021-02-04:
Dears,

I got an error (in below) when I wanted to use the PV to PQ conversion in pandapower as follows 
The main pice of code which got error:
"
import pandapower as pp
import pandapower.networks as nw
net = nw.case2848rte()

pp.runpp(net)
for i in net.gen.index:
    bus = net.gen.bus.at[i]
    p_mw = net.res_gen.p_mw.at[i]
    q_mvar = net.res_gen.q_mvar.at[i]
    in_service = net.gen.in_service.at[i]
    pp.create_sgen(net, bus=bus, p_mw=p_mw, q_mvar=q_mvar, in_service=in_service)
    net.gen.in_service.at[i] = False
pp.runpp(net, init="results")
"
The Error is related to the line in which I want to create sgen:
" ValueError: Cannot assign bool to float/integer series"

The error is only for RTE2848, IEEE24 and IEEE145 while for IEEE14, IEEE30, and IEEE57 it works very well.
I do not understand why for some networks it works but for some other networks it does not work.
Does it depends on the features of some networks?
Of course, I worked with this pice of code a few months ago in the other code and I did not have any problem. However, the previous code now gets this error for IEEE24, 145 and RTE2848.

Any comments would be appreciated.
Many thanks to you for your reply.

Best regards

wangzhenassd, 2021-02-04:
Hi, I tried your example with 2848rte but cannot reproduce the problem. Can you post your version of pandapower, pandas and numpy?

closed 2021-02-05 by DRL2019

DRL2019, 2021-02-05:
I solved the problem now (at 4:30 am CET, in the middle of my sleep :) ).
For unknown reason, pandapower accepts "net.sgen.controllable=1" for some networks mentioned in above and it does not accept for some other networks. But it accepts "net.sgen.controllable=True" for all mentioned networks in above.

DRL2019, 2021-02-05:
> Hi, I tried your example with 2848rte but cannot reproduce the problem. Can you post your version of pandapower, pandas and numpy?

Hi,

Sorry for delay.
The last version. This week I got this problem and I guessed that maybe because of the version of my installed pandapower. I installed again but the problem was not resolved.
As I commented last night I found that the "net.sgen.controllable=1" is not acceptable for some networks such as rte2848 and ieee24, 145 ...
Thanks for your reply.

Best regards