e2nIEE/pandapower #1135: Errors when using rundcpp in Pandapower to run DC power flow on islands during cascading failure simulation

crazystevepotatoes, 2021-02-08:
Hi guys! I am using the DC  power flow in Pandapower as a tool for cascading failure simulation. The idea of the simulation is to constantly track and delete overloaded lines once the failure is triggered. During the simulation several islands existed. For example, an island only has two buses and one transmission line connecting them.  Afterwards I implemented the DC power flow on the island which only has two buses and one line as shown in the first block.
`import pandapower as pdp`
`import pandapower.networks as pn`
`net = pn.GBreducednetwork()`
`pdp.drop_buses(net,list(range(1,27)))`
`net.gen.iloc[0, 0] =True`
`pdp.rundcpp(net)`
`
However, when I run the "rundcpp" command, errors are shown as follows.

`Traceback (most recent call last):
  File "C:\Users\hongminzhu\PycharmProjects\pythonProject\venv\lib\site-packages\IPython\core\interactiveshell.py", line 3418, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-38-caa47deedad0>", line 1, in <module>
    pdp.rundcpp(net)
  File "C:\Users\hongminzhu\PycharmProjects\pythonProject\venv\lib\site-packages\pandapower\run.py", line 274, in rundcpp
    _powerflow(net, **kwargs)
  File "C:\Users\hongminzhu\PycharmProjects\pythonProject\venv\lib\site-packages\pandapower\powerflow.py", line 62, in _powerflow
    ppc, ppci = _pd2ppc(net)
  File "C:\Users\hongminzhu\PycharmProjects\pythonProject\venv\lib\site-packages\pandapower\pd2ppc.py", line 57, in _pd2ppc
    net["_is_elements"] = aux._select_is_elements_numba(net)
  File "C:\Users\hongminzhu\PycharmProjects\pythonProject\venv\lib\site-packages\pandapower\auxiliary.py", line 449, in _select_is_elements_numba
    set_elements_oos(element_df["bus"].values, element_df["in_service"].values,
  File "C:\Users\hongminzhu\PycharmProjects\pythonProject\venv\lib\site-packages\numba\core\dispatcher.py", line 414, in _compile_for_args
    error_rewrite(e, 'typing')
  File "C:\Users\hongminzhu\PycharmProjects\pythonProject\venv\lib\site-packages\numba\core\dispatcher.py", line 357, in error_rewrite
    raise e.with_traceback(None)
numba.core.errors.TypingError: Failed in nopython mode pipeline (step: nopython frontend)
non-precise type array(pyobject, 1d, C)
During: typing of argument at C:\Users\hongminzhu\PycharmProjects\pythonProject\venv\lib\site-packages\pandapower\auxiliary.py (411)
File "venv\lib\site-packages\pandapower\auxiliary.py", line 411:
def _python_set_elements_oos(ti, tis, bis, lis):  # pragma: no cover
    for i in range(len(ti)):
    ^`
It seems that the error is related to the numba. Furthermore, I also did some trail-and-error work and found that this error usually comes with a network with only a few buses . I wonder the reason for the error and how to solve such problem. I would be very grateful if you could give me some feedbacks.

wangzhenassd, 2021-02-08:
Hi, can you try delete __pycache__ directory? It seems to be related to the numba compiled code.

crazystevepotatoes, 2021-02-08:
Sorry do you mean the pycache directory in numba?

crazystevepotatoes, 2021-02-08:
Sorry I didn't get your points.

wangzhenassd, 2021-02-08:
I main  `__pycache__` directory in your code, which is automatically generated

crazystevepotatoes, 2021-02-08:
I just tried that, it didn;t work.

wangzhenassd, 2021-02-08:
can you try the same code with numba uninstalled?

crazystevepotatoes, 2021-02-08:
`ERROR: Exception:
Traceback (most recent call last):
  File "c:\users\hongminzhu\pycharmprojects\pythonproject\venv\lib\site-packages\pip\_internal\cli\base_command.py", line 189, in _main
    status = self.run(options, args)
  File "c:\users\hongminzhu\pycharmprojects\pythonproject\venv\lib\site-packages\pip\_internal\commands\uninstall.py", line 91, in run
    uninstall_pathset.commit()
  File "c:\users\hongminzhu\pycharmprojects\pythonproject\venv\lib\site-packages\pip\_internal\req\req_uninstall.py", line 456, in commit
    self._moved_paths.commit()
  File "c:\users\hongminzhu\pycharmprojects\pythonproject\venv\lib\site-packages\pip\_internal\req\req_uninstall.py", line 296, in commit
    save_dir.cleanup()
  File "c:\users\hongminzhu\pycharmprojects\pythonproject\venv\lib\site-packages\pip\_internal\utils\temp_dir.py", line 205, in cleanup
    rmtree(ensure_text(self._path))
  File "c:\users\hongminzhu\pycharmprojects\pythonproject\venv\lib\site-packages\pip\_vendor\retrying.py", line 49, in wrapped_f
    return Retrying(*dargs, **dkw).call(f, *args, **kw)
  File "c:\users\hongminzhu\pycharmprojects\pythonproject\venv\lib\site-packages\pip\_vendor\retrying.py", line 212, in call
    raise attempt.get()
  File "c:\users\hongminzhu\pycharmprojects\pythonproject\venv\lib\site-packages\pip\_vendor\retrying.py", line 247, in get
    six.reraise(self.value[0], self.value[1], self.value[2])
  File "c:\users\hongminzhu\pycharmprojects\pythonproject\venv\lib\site-packages\pip\_vendor\six.py", line 703, in reraise
    raise value
  File "c:\users\hongminzhu\pycharmprojects\pythonproject\venv\lib\site-packages\pip\_vendor\retrying.py", line 200, in call
    attempt = Attempt(fn(*args, **kwargs), attempt_number, False)
  File "c:\users\hongminzhu\pycharmprojects\pythonproject\venv\lib\site-packages\pip\_internal\utils\misc.py", line 129, in rmtree
    shutil.rmtree(dir, ignore_errors=ignore_errors,
  File "C:\Users\hongminzhu\AppData\Local\Programs\Python\Python38\lib\shutil.py", line 737, in rmtree
    return _rmtree_unsafe(path, onerror)
  File "C:\Users\hongminzhu\AppData\Local\Programs\Python\Python38\lib\shutil.py", line 610, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "C:\Users\hongminzhu\AppData\Local\Programs\Python\Python38\lib\shutil.py", line 610, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "C:\Users\hongminzhu\AppData\Local\Programs\Python\Python38\lib\shutil.py", line 615, in _rmtree_unsafe
    onerror(os.unlink, fullname, sys.exc_info())
  File "C:\Users\hongminzhu\AppData\Local\Programs\Python\Python38\lib\shutil.py", line 613, in _rmtree_unsafe
    os.unlink(fullname)
PermissionError: [WinError 5] 拒绝访问。: 'C:\\Users\\hongminzhu\\PycharmProjects\\pythonProject\\venv\\Lib\\site-packages\\~umba\\core\\runtime\\_nrt_python.cp38-win_amd64.pyd'
`

crazystevepotatoes, 2021-02-08:
Did not work

wangzhenassd, 2021-02-08:
You need to completely close the python instance before the uninstallation

crazystevepotatoes, 2021-02-08:
`numba cannot be imported and numba functions are disabled.
Probably the execution is slow.
Please install numba to gain a massive speedup.
(or if you prefer slow execution, set the flag numba=False to avoid this warning!)

Traceback (most recent call last):
  File "C:/Users/hongminzhu/PycharmProjects/pythonProject/cascadingfailure/SOC/fast/modifiedfastpf.py", line 53, in <module>
    pdp.rundcpp(subnet)
  File "C:\Users\hongminzhu\PycharmProjects\pythonProject\venv\lib\site-packages\pandapower\run.py", line 274, in rundcpp
    _powerflow(net, **kwargs)
  File "C:\Users\hongminzhu\PycharmProjects\pythonProject\venv\lib\site-packages\pandapower\powerflow.py", line 62, in _powerflow
    ppc, ppci = _pd2ppc(net)
  File "C:\Users\hongminzhu\PycharmProjects\pythonProject\venv\lib\site-packages\pandapower\pd2ppc.py", line 57, in _pd2ppc
    net["_is_elements"] = aux._select_is_elements_numba(net)
  File "C:\Users\hongminzhu\PycharmProjects\pythonProject\venv\lib\site-packages\pandapower\auxiliary.py", line 449, in _select_is_elements_numba
    set_elements_oos(element_df["bus"].values, element_df["in_service"].values,
  File "C:\Users\hongminzhu\PycharmProjects\pythonProject\venv\lib\site-packages\pandapower\auxiliary.py", line 412, in _python_set_elements_oos
    if tis[i] and bis[ti[i]]:
IndexError: only integers, slices (`:`), ellipsis (`...`), numpy.newaxis (`None`) and integer or boolean arrays are valid indices`

wangzhenassd, 2021-02-08:
Ah ok, it is probably related to the network, when some internal array is empty. I tried your example with a try..except.. loop around this function and get this error, I think, this is the true cause of your problem.
`UserWarning: No reference bus is available. Either add an ext_grid or a gen with slack=True`

wangzhenassd, 2021-02-08:
```python
net.gen.in_service.iloc[0] =True
```

The problem is on your test code with `net.gen.iloc[0, 0] =True` , which changed the type of data, which than caused the problem in numba.

crazystevepotatoes, 2021-02-08:
I really appreciate your help. However, I am not sure if the setting of slack bus is the key to this problem. The shor codes I posted is a minimal example of my problem. However, I already set the slack bus for each island but the code stiil doesn't work.

crazystevepotatoes, 2021-02-08:
`import numpy as np
import pandas as pd
import pandapower as pdp
import pandapower.networks as pn
from cascadingfailure.SOC.fast.islands import islands
from cascadingfailure.SOC.fast.droplist import droplist
from pandapower.plotting import simple_plot
import matplotlib.pyplot as plt
import copy
import random
from pytictoc import TicToc

net_0 = pn.GBreducednetwork()
net = copy.deepcopy(net_0)
net_2 = copy.deepcopy(net_0)
thres = 80
pdp.rundcpp(net_2)
cap_ratio = 1.2
capacities = net_2.res_line.iloc[:, 0].apply(np.abs) * cap_ratio
random.seed()
new_failed_lines = []
# first_line = random.choice(net_2.line.index.tolist())
first_line = 56
print(first_line)
new_failed_lines.append(first_line)
print(net_0.line.shape)

# use power loading rather than the previously used line loading percent as a metric
while len(new_failed_lines) > 0:
    pdp.drop_lines(net, new_failed_lines)
    pdp.rundcpp(net)

    if net.res_line.i_ka.isnull().sum() > 0:
        new_failed_lines = []
        _, _, list_single_bus = islands(net)
        if list_single_bus:
            pdp.drop_buses(net, list_single_bus)
            subG, dict_subg_nodes, list_single_bus = islands(net)
            all_bus = net.bus.index.tolist()
        else:
            list_single_bus = []
            subG, dict_subg_nodes, list_single_bus = islands(net)
            all_bus = net.bus.index.tolist()
        for i in list(dict_subg_nodes.keys()):
            subnodes = dict_subg_nodes[i]
            print('///////////////////')
            drop_nodes = droplist(all_bus, subnodes)
            subnet = copy.deepcopy(net)
            pdp.drop_buses(subnet, drop_nodes, drop_elements=True)
            print(drop_nodes)
            subnet.gen.loc[0, 'slack'] = True
            print(subnet)
            pdp.rundcpp(subnet)
            line_power_flow = subnet.res_line.iloc[:, 0]
            loading = pd.Series(index=line_power_flow.index, name='loading', dtype='float64')
            for j in line_power_flow.index.tolist():
                loading.loc[j] = abs(line_power_flow.loc[j]) / capacities.loc[j] * 100
            print(loading)
            assert line_power_flow.shape == loading.shape
            for j in line_power_flow.index.tolist():
                if loading.loc[j] >= thres:
                    new_failed_lines.append(j)
            if loading.max() < thres:
                new_failed_lines = []
                print('Simulation stopped')

        assert len(new_failed_lines) == len(list(set(new_failed_lines)))

    elif net.res_line.i_ka.isnull().sum() == 0:
        net.gen.loc[0, 'slack'] = True
        new_failed_lines = []
        line_power_flow = net.res_line.iloc[:, 0]
        loading = pd.Series(index=line_power_flow.index, name='loading', dtype='float64')
        for j in line_power_flow.index.tolist():
            loading.loc[j] = abs(line_power_flow.loc[j]) / capacities.loc[j] * 100
        print(loading)
        assert line_power_flow.shape == loading.shape
        for j in line_power_flow.index.tolist():
            if loading.loc[j] > thres:
                new_failed_lines.append(j)
        if loading.max() < thres:
            new_failed_lines = []
            print('Simulation stopped')
a = net.res_load.loc[:, 'p_mw'].sum()
b = net_2.res_load.loc[:, 'p_mw'].sum()
print((b-a)/b)
`

crazystevepotatoes, 2021-02-08:
This is my complete code. You can replicate the error based on this.

crazystevepotatoes, 2021-02-08:
import networkx as nx
import pandapower.topology as top

def islands(net):
    list_single_bus = []
    G = top.create_nxgraph(net, respect_switches=False, include_trafos=True, multi=True)
    subG = [G.subgraph(c) for c in nx.connected_components(G)]
    if len(subG) == 1:
        dict_subg_nodes = {}
        subgraph = subG[0] # multigraph
        print('No islands.')
        nodes = subgraph.nodes()
        nnodes = len(nodes)
        edges = subgraph.edges()
        nedges = len(edges)
        print('The network contains {} nodes and {} edges'.format(nnodes, nedges))
        dict_subg_nodes[0] = list(nodes)
    elif len(subG) > 1:
        dict_subg_nodes = {}
        print('several islands exist.')
        for i in range(len(subG)):
            subgraph = subG[i] # multigraph
            nodes = subgraph.nodes()
            nnodes = len(nodes)
            edges = subgraph.edges()
            nedges = len(edges)
            dict_subg_nodes[i] = list(nodes)
            print('The island {} has {} nodes and {} edges'.format(i+1,nnodes, nedges))
            if nnodes == 1:
                print('single bus exists.')
                list_single_bus.append(list(nodes)[0])
    if list_single_bus:
        print(list_single_bus)

    return subG, dict_subg_nodes, list_single_bus

crazystevepotatoes, 2021-02-08:
def droplist(list1, list2):
    # drop elments of list2 in list1
    assert len(list1) >= len(list2)
    a = list1
    for i in range(len(list2)):
        if list2[i] in list1:
            a.remove(list2[i])
    return a

wangzhenassd, 2021-02-08:
Can you reformulate the code? It seems a little bit strange.

crazystevepotatoes, 2021-02-08:
yeah

crazystevepotatoes, 2021-02-08:
I'll send an .zip file

crazystevepotatoes, 2021-02-08:
[zipcodes.zip](https://github.com/e2nIEE/pandapower/files/5943001/zipcodes.zip)

wangzhenassd, 2021-02-08:
```python
# subnet.gen.loc[0, 'slack'] = True
subnet.gen.slack.iloc[0] = True```

I think it is a pandas issue, which changed the data type internally. By doing this adaption the problem with numba can be avoided.

crazystevepotatoes, 2021-02-08:
Nice!!! It works. Many thanks for your help!

wangzhenassd, 2021-02-08:
You are welcome. I will close the issue, since there is no further issues regarding numba implementation.

closed 2021-02-08 by wangzhenassd