e2nIEE/pandapower #1138: Issue with hosting capacity (visualization of results)

NontasPapapadopoulos, 2021-02-08:
Hello gentlemen, 

I`m trying to run a simulation of hosting capacity, using the provided algorithm from https://github.com/e2nIEE/pandapower/blob/develop/tutorials/hosting_capacity.ipynb 

Unfortunately, i get this error, it has to do with Seaborn: 
I searched for the solution but i could not solve it. I`m wondering if anyone else had the same issue. 

C:\Users\Detonator\PycharmProjects\Converter\venv\lib\site-packages\**seaborn**\_decorators.py:36: FutureWarning: Pass the following variable as a keyword arg: x. From version 0.12, the only valid positional argument will be `data`, and passing other arguments without an explicit keyword will result in an error or misinterpretation.
  warnings.warn(

C:\Users\Detonator\PycharmProjects\Converter\venv\lib\site-packages\**seaborn**\_core.py:1303: UserWarning: Vertical orientation ignored with only `x` specified.
  warnings.warn(single_var_warning.format("Vertical", "x"))

C:/Users/Detonator/PycharmProjects/Converter/hosting_capacity.py:72: UserWarning: FixedFormatter should only be used together with FixedLocator
  **ax.set_xticklabels([""])** 

[image]

[image]

thanks in advance!

rbolgaryn, 2021-02-17:
Hi @NontasPapapadopoulos ,

seems like we will need to update the tutorial so that it works with the latest version of seaborn.

jwiemer112, 2023-01-10:
Is this still a thing? @NontasPapapadopoulos I get zero Warnings if I execute the tutorial.