e2nIEE/pandapower #1139: Warnings when using PowerModels.jl

andrea-belle, 2021-02-08:
Hello,

I am currently using the interface with PowerModels.jl, and I am facing some problems with the warnings. I am using Python in Anaconda. As I already mentioned in #1020 , when calling PowerModels.jl iteratively in Spyder you get an increasing loop of warnings. When using the Anaconda PowerShell instead of Spyder, you get one warning per iteration, which is much better. However, when using PowerModels.jl with the Anaconda Shell, I get a list of info and warnings regarding the generators costs, something like this (this is just an extract of the warnings):

`[warn | PowerModels]: active generators found at bus 353, updating to bus type from 1 to 2`
`[warn | PowerModels]: active generators found at bus 232, updating to bus type from 1 to 2`
`[warn | PowerModels]: active generators found at bus 153, updating to bus type from 1 to 2`
`[info | PowerModels]: removing 1 cost terms from generator 306: Any[100.0, -0.0]`
`[info | PowerModels]: removing 1 cost terms from generator 407: Any[100.0, -0.0]`
`[info | PowerModels]: removing 1 cost terms from generator 1: Any[1.0, 0.0]`

Is there a way to suppress them? Printing them every time slows down the simulation considerably.

Thanks in advance.

Andrea

rbolgaryn, 2021-06-20:
Hi @andrea-belle ,

there is a package (that is still under development) PandaModels.jl

It implements the interface to PowerModels in a better way, and there is no warning if you use it.

Once it is fully tested, we will switch to it as the only way to use PowerModels with pandapower.

Please check it out and let us know if it works for your use-case.

Roman

MMajidi137, 2021-06-20:
> Hi @andrea-belle ,
>
> there is a package (that is still under development) PandaModels.jl
>
> It implements the interface to PowerModels in a better way, and there is no warning if you use it.
>
> Once it [...]

Dear @andrea-belle, 

Also as a temporary solution, you may add 
`PowerModels.silence()`
before `function run_powermodels(json_path)` in script: pandapower/opf/run_powermodels.jl

closed 2021-07-21 by friederikemeier