e2nIEE/pandapower #1140: simplifying "create simple" exampe network leads to too high reactive power

Marahop, 2021-02-09:
Hi everyone,

I adjusted the sample from https://github.com/e2nIEE/pandapower/blob/master/tutorials/create_simple.ipynb
so there is just one static generator (sgen) and the external grid and the transformator.
Could you give me some advice please on why the reactive power here is higher than the power in Line 1 (line 0)? What have I done wrong? I just deleted other generator, load and a lines and busses..

p_from_mw | q_from_mvar 
-1.955356 | -7.154292

Here is my code:
https://github.com/Marahop/easyflow/blob/main/1strang_Netzwerk.py

Thank you very much.

rbolgaryn, 2021-02-17:
Hi @Marahop , I am getting a "Page not found" error

rbolgaryn, 2021-03-17:
Hi @Marahop ,

can you post your code here?

Roman

SteffenMeinecke, 2021-11-11:
closed due to inactivity

closed 2021-11-11 by SteffenMeinecke