e2nIEE/pandapower #1141: Dispatchable load with constant cos_phi not usable

Davere, 2021-02-10:
I use PP Version 2.5.0, I am currently going through the pandapower/pypower opf process because I have problems with the convergence. In the opf_setup line 164 creates the subfunction makeAvl additional constraints for dispatchable loads, which are identified by gens, which have pmax = 0 and pmin < 0. Additionaly it is required to have exactly one qmax/qmin value set and the other one need to be 0. (found in makeAvl.py line 49)
I tried to create a dispatchable load, but my q value which was 0 is overwritten by the initialisation in build_gen.py --> _build_pp_pq_element -->add_q_constraints (line 261) in this subfunction is a delta added/substracted to the qmax/qmin values, this value is defined in the net._options["delta"]. But I don`t think, that its the best way to set the delta = 0.
So my first idea was to add/substract the delta when checking the q == 0.
I just realize, the same problem is with checking the pmax == 0.

friederikemeier, 2021-07-21:
@rbolgaryn I won't be able to deal with this issue, sorry.

rbolgaryn, 2021-07-22:
Hi @Davere,

Can you write a test for this?

Roman

vogt31337, 2025-03-31:
closing due to inactivity.