e2nIEE/pandapower #1146: pandapower doesn't seem to work with Numpy 1.20 (latest)

rayanelhelou, 2021-02-18:
_First off, thank you for this amazing tool. Pandapower is great!!_ Here's my issue:

After upgrading Numpy to 1.20, I can no longer run powerflow (`pp.runpp(net)`). To try to prove that the issue is with numpy, I rolled back to numpy 1.19.0, and it worked like a charm. In fact, when I run `pandapower.test.run_all_tests()`, it mentions a failure related to numpy deprecations.

Opinion: It's possibly because of how datatypes are used in pandapower. For example, as mentioned in the following link, `np.int` can no longer be used; rather, you should just use `int`. [Numpy 1.20 Deprecations](https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations).

If I'm missing something very basic or obvious, please let me know.

wangzhenassd, 2021-02-18:
Yes, you are right, it seems that numpy 1.2.0 caused some problem with the bfsw powerflow algorithm and also the tests on github fail. 
@rbolgaryn Since it might need sometime to fix the bfsw, can we firstly set the numpy version below 1.2.0 in dependency?

rbolgaryn, 2021-02-22:
Hi @rayanelhelou , 

thank you for pointing it out. Indeed, it seems that the new release of numpy has broken some things in pandapower.
As you suggested @wangzhenassd, we will set the requirement below 1.2.0 for now until we fix it.

Roman

rbolgaryn, 2021-02-22:
It actually seems to be a scipy issue. Tests work with numpy 1.20.1 but don't work with scipy 1.6.1.
I limited scipy to 1.6.0 and we will remove the limit after fixing the underlying issue.

rbolgaryn, 2021-02-22:
Hi @jkrstulo , can you please take a look at this issue?

rayanelhelou, 2021-03-22:
Is there any update on this?

I tried working in my same venv (where all my other important packages are), and even when I upgraded pandapower it didn't work. However, when I work in a whole new venv, latest pandapower works well with latest numpy. Can you guess what might cause this? I'm trying to not have to reinstall all my other packages.

QasimWani, 2021-03-29:
yup. there's some issues with v >1.20. i fixed mine by
downgrading `pandapower` to version `<1.20`
```python
import sys
!$sys.executable -m pip install -U "numpy<1.20"
```

rbolgaryn, 2021-06-14:
Hi @rayanelhelou ,

unfortunately we still have no solution at the moment, still the only way to avoid the error is to downgrade numpy.

Roman

commit 942a16e4 (2022-01-10): fix numpy import; close #1146
closed 2022-01-10 by SteffenMeinecke