e2nIEE/pandapower #1147: Ineffective Capacitor Shunt connected to ext_grid

DRL2019, 2021-02-19:
Dears,

I'd like to study the influence of capacitor banks on the network in pandapower. I created a capacitor shunt connected to the ext_grid and then changed its reactive power, but it did not affect on the voltage of other buses. Here is a piece of my code, for instance, for IEEE14 & 57:
[CBank_ext_grid_test.txt](https://github.com/e2nIEE/pandapower/files/6009548/CBank_ext_grid_test.txt)

The code works very well by changing the reactive power of other capacitor shunts of IEEE14,57.
Is there any problem to create capacitor bank or I missed some conceptions ...?
Of course, I defined capacitor shunts for non-slack buses for other networks like CIGRE and the code worked very well.

Any comments would be appreciated in advance.

Best regards,

rbolgaryn, 2021-02-23:
Hi @DRL2019 ,

do I understand correctly that you are connecting a shunt capacitor to the same bus as the ext grid? 
And that the voltages at other buses are not influenced by this?
The ext_grid element is connected to a slack bus, so the voltage magnitude and voltage angle are defined by ext_grid. The resulting active and reactive power at the slack bus is calculated to match the set voltage angle and magnitude.
Whatever active or reactive power you additionally inject or consume at the slack bus, it will have no effect on the rest of the grid.

Roman

closed 2021-02-23 by rbolgaryn

DRL2019, 2021-02-23:
Hi @rbolgaryn 

Many thanks for your reply. 
Yes, you correctly understood my question.

As you know, slack bus is used to balance the active power |P| and reactive power |Q| in a electric network while performing load flow studies. So, the slack bus is used to provide for network losses by emitting or absorbing active and/or reactive power to and from the network. 

As a result, I expect that any change in the slack bus reactive power, through connecting a capacitor bank to it and changing its reactive power, the injected reactive power to slack bus, in parallel to keep fixed its voltage magnitude and angle, is also used to balace the reactive power of the network and thus it affects the voltage of load buses.
Am I wrong?

Thanks for your reply in advances.

Best regards,