e2nIEE/pandapower #1148: plot_voltage_profile

SimonRubenDrauz, 2021-02-19:
I wanted to plot a voltage profile of a meshed grid in pandapower and used the simbench mixed HV grid for that. 

The function I used can be found in plotting: plot_voltage_profile.

The result seems a bit confusing and might be an error:
[image]

As you can see some ext_grids are not beginning at distance 0. Furtermore, some shape seem to be duplicated. 

Looking forward hearing from you on this topic.

rbolgaryn, 2021-02-24:
Having them all start at the same point in a meshed grid is not necessarily desired, in my opinion

SimonRubenDrauz, 2021-02-24:
This is something, what needs to be discussed in general. I totally agree, I was also thinking that it might make sense that only one ext_grid is considered as reference and the others are analysed in contrast to that. However the solution right now seems a bit confusing to me. E.g. some shapes appear several times. I think, what happens is, that in case of a closed loop we start at one ext grid and end up at the particular ext_grid again, however, not at point zero but for example at 75 km (all lines lengths of one loop summed up). I think, this might lead to confustion, what do you think?

SimonRubenDrauz, 2021-02-24:
Furthermore, in the current solution you cannot see anymore that you have closed loops. You see, that ext grids are connected to each other, but you cannot see that some parts of the grid are arranged in a ring.

SteffenMeinecke, 2021-11-11:
I would say both solution are desirable in principle:

1. one slack is fixed to distance=0, all other buses are relative to that slack bus
2. all slack buses are fixed to distance=0, all other buses are located at the minimum distance to the slack buses

In both solutions the docstrings of the function should note that the shapes do not necessarily represent the line length.
In both solutions, lines should not occur multiple times as shapes.

SimonRubenDrauz, 2021-11-11:
I agree with you, however, I still think in the current version there is some kind of bug. It seems that the same slack is two times plotted at different distances. This can be seen as some shapes do repeat:
[image]
Here you see the voltage profile for a slack with position 0. The exact same behaviour can be seen at positon 150:
[image]
So it seems one and the same slack is plotted twice at two different locations. It is not that another slack is plotted relatively to the first one.