e2nIEE/pandapower #1157: Difference between the type of some buses in Pandapower & Pypower

DRL2019, 2021-03-01:
Dears,

When working with some networks in Pandapower, thanks to the pandapower developers, I noticed that there are some buses that are known as load buses in pypower case (ppc) but they are not as load/gen/sgen/slack/shunt buses in pandapower.
For example, I checked that in RTE2848, there are 979 buses which are not as slack/load/gen/sgen/shunt buses while they are as load buses in ppc.
Why is there this difference?

Thanks for your reply in advance.

Best regards,

rbolgaryn, 2021-03-02:
Hi @SteffenMeinecke , can you please take a look?

SteffenMeinecke, 2021-03-02:
Dear DRL2019,
please extract the attached testfiles and have a look. With the attached python script I determined 979 of such buses - also with the matpower case. Please specify your question if this answer is not sufficient.

[testfiles.zip](https://github.com/e2nIEE/pandapower/files/6069897/testfiles.zip)

DRL2019, 2021-03-02:
Dear @SteffenMeinecke,

Many thanks for your reply and sending the file.
Actually, I noticed that the similar issue when working on the other networks like IEEE14, 57, 118. Let me use the script that you sent and come back to you if there is any question. 
Once more, thanks to you and the other Pandapower developers.

Best regards,

DRL2019, 2021-03-03:
Dear @SteffenMeinecke ,

I saw the python script that is very good to find a set of buses which are not load/gen/sgen/shunt/slack in pandapower. That is helpful to recognise such buses. Thank you very much 🙏🙏🙏
Actually, the question has been "why" such buses are load buses in ppc but in pandapower they are not in this category. Is there any concept that I miss in the context?
Any comment would be appreciated.

Best regards,

SteffenMeinecke, 2021-03-03:
I am sorry but I don't know which buses (buses that are load buses in ppc but not in pandapower) you mean.
Didn't I show that in RTE2848 ppc and pandapower have the same number of buses which are not connected to any of loads, gens, shunts, ... ?
The pandapower networks from IEEE are converted from ppc/mpc using the [pypower converter](https://github.com/e2nIEE/pandapower/blob/develop/pandapower/converter/pypower/from_ppc.py). You may have a look in there to understand how it works. There is no concept to change the network in such a way. Only converter issues because a difference in the pandapower and ppc data structure.

DRL2019, 2021-03-03:
Dear @SteffenMeinecke 

Thank you for your complete and helpful explanations.
Actually, when I looked at ppc buses excel file, created by
DataFrame(net._ppc["internal"]["bus"]).to_excel("bus_ppc.rte2848.xlsx"),
 I saw in the third column the types of some buses (979 buses) of RTE2848 have code 1 (load bus code in ppc).
On the other hand, when I looked at the load buses excel file, created by
DataFrame(net.load).to_excel("pp_load.rte2848.xlsx),
I noticed that there are just 1388 load buses in RTE2848 in pandapower. When searching those 979 buses, there are not in the net.load.

I thought I might miss some concepts. I did not know so I prefered to ask you.
I attached both excel files here:

[bus_ppc.rte2848.xlsx](https://github.com/e2nIEE/pandapower/files/6077355/bus_ppc.rte2848.xlsx)

[pp_load.rte2848.xlsx](https://github.com/e2nIEE/pandapower/files/6077364/pp_load.rte2848.xlsx)

Many thanks for your consideration and time in advance.

Best regards,

SteffenMeinecke, 2021-03-04:
In `ppc["bus"][:, 1]` `1` is the default value. Thus not only the buses have the value `1`, to which also a load is connected, but also all other buses, which are not PV buses (`net.gen`) or Slack buses (`net.ext_grid`).

DRL2019, 2021-03-05:
Thanks for your useful comments. Now, everything is clear for me. The number "1" in the "type" column of the ppc bus table (created by net._ppc["internal"]["bus"]) does not necessarily mean the bus is a liad bus.
Thanks a lot.
So I close this issue.
Best regards,

closed 2021-03-05 by DRL2019