e2nIEE/pandapower #1158: Difference between MATPOWER case 30 and pandapower case 30

eendru, 2021-03-02:
Dears, i have some misunderstanding between two cases: [Matpower case 30](https://matpower.org/docs/ref/matpower5.0/case30.html) and [Pandapower case 30](https://pandapower.readthedocs.io/en/v2.5.0/networks/power_system_test_cases.html?highlight=case14#pandapower.networks.case_ieee30)

Take a look to generators:
Matpower: 
`
0063 %% generator data
0064 %    bus    Pg    Qg    Qmax    Qmin    Vg    mBase    status    Pmax    Pmin    Pc1    Pc2    Qc1min    Qc1max    Qc2min    Qc2max    ramp_agc    ramp_10    ramp_30    ramp_q    apf
0065 mpc.gen = [
0066     1    23.54    0    150    -20    1    100    1    80    0    0    0    0    0    0    0    0    0    0    0    0;
0067     2    60.97    0    60    -20    1    100    1    80    0    0    0    0    0    0    0    0    0    0    0    0;
0068     22    21.59    0    62.5    -15    1    100    1    50    0    0    0    0    0    0    0    0    0    0    0    0;
0069     27    26.91    0    48.7    -15    1    100    1    55    0    0    0    0    0    0    0    0    0    0    0    0;
0070     23    19.2    0    40    -10    1    100    1    30    0    0    0    0    0    0    0    0    0    0    0    0;
0071     13    37    0    44.7    -15    1    100    1    40    0    0    0    0    0    0    0    0    0    0    0    0;
0072 ];
`

Pandapower
`
bus | controllable | in_service | name | p_mw | scaling | sn_mva | type | vm_pu | slack | max_p_mw | min_p_mw | max_q_mvar | min_q_mvar
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
1 | True | True | None | 60.97 | 1.0 | nan | None | 1.0 | False | 80.0 | 0.0 | 60.0 | -20.0
21 | True | True | None | 21.59 | 1.0 | nan | None | 1.0 | False | 50.0 | 0.0 | 62.5 | -15.0
26 | True | True | None | 26.91 | 1.0 | nan | None | 1.0 | False | 55.0 | 0.0 | 48.7 | -15.0
22 | True | True | None | 19.2 | 1.0 | nan | None | 1.0 | False | 30.0 | 0.0 | 40.0 | -10.0
12 | True | True | None | 37.0 | 1.0 | nan | None | 1.0 | False | 40.0 | 0.0 | 44.7 | -15.0
`

There isn't generator with bus_id = 1 (in original Matpowers case, and p_mw = 23.54) 

What am I missing?

wangzhenassd, 2021-03-02:
Hi,
I think gen 1 in case30 from matpower is modelled as ext_grid (slack) in pandapower

eendru, 2021-03-02:
Yeah, thank you.
And I can obtain it from `res_ext_grid`

closed 2021-03-02 by eendru