e2nIEE/pandapower #1160: TCSC in pandapower

lschewe, 2021-03-08:
We want to model a power grid that contains a TCSC (something like [this](https://library.e.abb.com/public/fdf0b019e1fe08a48325771f002dbfc5/A02-0158.pdf)). We do not see how to model this in PP and would be happy for any advice.

jkisse, 2021-03-08:
This element should also be an active part in OPF calculations.

lthurner, 2021-03-08:
I think for power flow this could be modeled with a controller that controls an impedance element.

However, you mentioned that this should be an OPF element. Can you describe how this element should behave, i.e. what would be the optimization variable and constraints?

rbolgaryn, 2021-07-07:
Dear @lschewe ,

have you tried implementing the functionality of the device in a controller? Has it been a viable option for you?

Roman

closed 2021-07-07 by rbolgaryn