e2nIEE/pandapower #1163: Multiple external girds

AlemsegedWeldeyesus, 2021-03-09:
We want to model a network that involves many transmission lines and multiple external grids, and then run OPF. Is having multiple external grids still an issue?

rbolgaryn, 2021-03-10:
Hi @AlemsegedWeldeyesus ,

pandapower supports having multiple ext_grid elements in the grid.

Also, take a look at #771 

Roman

lthurner, 2021-03-11:
In power flow calculation, it is definetly supported and well tested.

In OPF, it should work as well, but I am not sure if it es thouroughly tested yet. Did you try to put together a minimal example to check if it works @AlemsegedWeldeyesus ?

AlemsegedWeldeyesus, 2021-03-12:
I have tried to put a minimal example. The power flow calculation is fine. But when I try to run OPF, I get the following message at the end. "Multiple connected slacks in one area found. This would probably lead to non-convergence of the OPF. I'll change all but one slack (ext_grid) to gens. To avoid undesired behaviour, rather convert the slacks to gens yourself and set slack=True for one of them."

rbolgaryn, 2021-03-16:
have you tried converting all slacks to gen elements with the parameter slack=True?

rbolgaryn, 2021-07-07:
Closing the issue due to inactivity.

@AlemsegedWeldeyesus , please reopen if you have any updates about the issue.

Roman

closed 2021-07-07 by rbolgaryn

PB-vines, 2023-10-07:
RE even if you change the external grids to gens with slack = True, you get the same warning.