e2nIEE/pandapower #1167: using and modifying benchmark network

Marahop, 2021-03-16:
Hi, 
I would like to work with a benchmark network like  the following:
Medium voltage distribution network
net = pn.create_cigre_network_mv(with_der=False)

I would like to change a few things like sgen without creating the same network from scratch. Is this possible and if so, how?
Thank you very much! :)

rbolgaryn, 2021-03-17:
Hi @Marahop ,

it is possible to adjust the network after you load it.
For example, you can create additional static generators like this:

```python
import pandapower as pp
import pandapower.networks as pn

net = pn.create_cigre_network_mv(with_der=False)
pp.create_sgen(net, bus=1, p_mw=0.5, q_mvar=0, name='Example')
```

You can write your own function, in which the grid is first loaded from pp.networks and then modified according to your use-case.

Best regards, 
Roman

Marahop, 2021-03-17:
Hi Roman, 
thank you so much for your response.
`import pandapower as pp
import pandapower.networks as pn

net = pn.create_cigre_network_mv(with_der=False)
pp.create_load(net, bus=1;p_mw=0.8, q_mvar=0)`

The last line adds a load at bus one. What I need is to modify/change the load at bus one, the one that already exists in the network, which is originally p_mw= 14.9 and called "Load R1".

so I thought perhaps this would work, but it didnt : `pp.create_load(net, bus=1;p_mw=0.8, q_mvar=0, name=''Load R1)`

Happy for any advice :) Thank you!!

rbolgaryn, 2021-03-17:
You can modify the values by using pandas, e.g. net.load.at[1, "p_mw"] = 1.5

Marahop, 2021-03-17:
Hi rbolgaryn! That is great, that is what I was looking for. Works wonderfully :)!! so happy here, couldnt find this nowhere! THX

closed 2021-04-29 by rbolgaryn