e2nIEE/pandapower #1172: Question: Is there a way to directly handle the results after using the Timeseries tool and not write them to the hard disk?

Crueda96, 2021-03-19:


TOffergeld, 2021-03-23:
I've accessed the result data in the `OutputWriter` table of the net when I needed to: 
`net.output_writer.object[0]["res_bus.vm_pu"]`, for the zero-th `OutputWriter`. If you have multiple `OutputWriter`s adjust accordingly. But a documented and more intuitive way would probably be helpful.

Crueda96, 2021-03-23:
@TOffergeld Thank you very much for the information, maybe you can help me when I apply the recommended structure: `net.output_writer.object[0]['res_bus.vm_pu']` ,I get an error message: `TypeError: 'OutputWriter' object is not subscriptable`.
Thanks in advance

TOffergeld, 2021-03-23:
Sorry, it's `net.output_writer.object[0].output["res_bus.vm_pu"]`.

closed 2021-03-29 by Crueda96