e2nIEE/pandapower #1175: Documentation: Correct signing of reactive power in load element

rockstaedt, 2021-03-23:
Hello!

I just added a new load to the minimal example jupyter notebook to investigate the signing of reactive power. Apparently, a positive 'q_mvar' value leads to a voltage decrease and hence an inductive behavior. In the documentation (v 2.6.0), this is stated the other way around: "reactive power will be negative for inductive behavior".

Can someone verify this?

Load was added by: 

`pp.create_load(net, bus=bus3, p_mw=0, q_mvar=0.5, name="testL")`

And with a negative value respectively. 

Cheers.

rbolgaryn, 2021-03-30:
Hi @rockstaedt ,

thank you for noticing it, it is indeed an error in the documentation. 
We will adjust the documentation soon.

Best regards, 
Roman

rockstaedt, 2021-03-31:
Hi @mkraiczy!
Yes, absolutely! Thanks for this extensive answer 🙃

closed 2021-06-15 by rbolgaryn