e2nIEE/pandapower #1178: Unwanted error in the OPF AC when trying to combine controllable out of service and non controllable in service

Crueda96, 2021-03-25:
As mentioned in issue#926 , an unwanted error is also generated in the OPF AC when trying to combine controllable out of service and non controllable in service 

Minimal Example:

`import pandapower as pp
net = pp.create_empty_network()
bus = pp.create_bus(net, vn_kv=110., name="Bus 2")
pp.create_ext_grid(net, bus=bus, vm_pu=1.00, name="Grid Connection")
pp.create_gen(net, bus=bus, p_mw=5, min_p_mw = 5, max_p_mw = 20, name='gen 1')
pp.create_gen(net, bus=bus, p_mw=5, min_p_mw = 5, max_p_mw = 20, name='gen 3')

net.gen[['in_service', 'controllable']] = [[False, True],
                                           [True, False]]

pp.runopp(net,verbose = True)`

Output: 

`---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
<ipython-input-37-47fe18b0aa9e> in <module>
      9                                            [True, False]]
     10 
---> 11 pp.runopp(net,verbose = True)

C:\Users\Daniel Rueda\.conda\envs\Tesis\lib\site-packages\pandapower\run.py in runopp(net, verbose, calculate_voltage_angles, check_connectivity, suppress_warnings, switch_rx_ratio, delta, init, numba, trafo3w_losses, consider_line_temperature, **kwargs)
    355     _check_bus_index_and_print_warning_if_high(net)
    356     _check_gen_index_and_print_warning_if_high(net)
--> 357     _optimal_powerflow(net, verbose, suppress_warnings, **kwargs)
    358 
    359 

C:\Users\Daniel Rueda\.conda\envs\Tesis\lib\site-packages\pandapower\optimal_powerflow.py in _optimal_powerflow(net, verbose, suppress_warnings, **kwargs)
     39     init_results(net, "opf")
     40 
---> 41     ppc, ppci = _pd2ppc(net)
     42 
     43     if not ac:

C:\Users\Daniel Rueda\.conda\envs\Tesis\lib\site-packages\pandapower\pd2ppc.py in _pd2ppc(net, sequence)
    113         aux._set_isolated_buses_out_of_service(net, ppc)
    114 
--> 115     _build_gen_ppc(net, ppc)
    116 
    117     if "pf" in mode:

C:\Users\Daniel Rueda\.conda\envs\Tesis\lib\site-packages\pandapower\build_gen.py in _build_gen_ppc(net, ppc)
     51     _init_ppc_gen(net, ppc, f)
     52     for element, (f, t) in gen_order.items():
---> 53         add_element_to_gen(net, ppc, element, f, t)
     54     net._gen_order = gen_order
     55 

C:\Users\Daniel Rueda\.conda\envs\Tesis\lib\site-packages\pandapower\build_gen.py in add_element_to_gen(net, ppc, element, f, t)
     80         _build_pp_ext_grid(net, ppc, f, t)
     81     elif element == "gen":
---> 82         _build_pp_gen(net, ppc, f, t)
     83     elif element == "sgen_controllable":
     84         _build_pp_pq_element(net, ppc, "sgen", f, t)

C:\Users\Daniel Rueda\.conda\envs\Tesis\lib\site-packages\pandapower\build_gen.py in _build_pp_gen(net, ppc, f, t)
    220         ppc = _check_gen_vm_limits(net, ppc, gen_buses, gen_is)
    221         if "controllable" in net.gen.columns:
--> 222             ppc = _enforce_controllable_vm_pu_p_mw(net, ppc, gen_is, f, t)
    223 
    224 

C:\Users\Daniel Rueda\.conda\envs\Tesis\lib\site-packages\pandapower\build_gen.py in _enforce_controllable_vm_pu_p_mw(net, ppc, gen_is, f, t)
    184     # if there are some non controllable gens -> set vm_pu and p_mw fixed
    185     if np.any(not_controllable):
--> 186         bus = net["gen"]["bus"].values[not_controllable]
    187         vm_pu = net["gen"]["vm_pu"].values[not_controllable]
    188         p_mw = net["gen"]["p_mw"].values[not_controllable]

IndexError: boolean index did not match indexed array along dimension 0; dimension is 2 but corresponding boolean dimension is 1`

rbolgaryn, 2021-06-20:
Hi @Crueda96 ,

thank you for the minimal example, we will have a look.

Roman

friederikemeier, 2021-07-21:
Hi, I agree that this is an unwanted behaviour. I won't have the time to fix this myself, but a pull request on this would highly be appreciated!

friederikemeier, 2021-07-21:
This is related to #926 where the same bug is reported.

vogt31337, 2026-11-12:
When running the above code an error is generated:
```console
These columns are missing in gen: <StringArray>
['min_q_mvar', 'max_q_mvar']
Length: 2, dtype: str

KeyError: 'OPF parameters are not set correctly. See error log.'
```

The min and max reactive power for the two gens are missing. Adding solves the problem and the opf runs:
```python
import pandapower as pp
net = pp.pandapowerNet(name="test")
bus = pp.create_bus(net, vn_kv=110., name="Bus 2")
pp.create_ext_grid(net, bus=bus, vm_pu=1.00, name="Grid Connection")
pp.create_gen(net, bus=bus, p_mw=5, min_p_mw = 5, max_p_mw = 20, name='gen 1', min_q_mvar=0, max_q_mvar=20)
pp.create_gen(net, bus=bus, p_mw=5, min_p_mw = 5, max_p_mw = 20, name='gen 3', min_q_mvar=0, max_q_mvar=20)

net.gen[['in_service', 'controllable']] = [[False, True],
                                           [True, False]]

pp.runopp(net,verbose = False)
```

Output: `no costs are given - overall generated power is minimized`

closed 2026-11-12 by vogt31337