e2nIEE/pandapower #1179: Over load and Wrong switch configuration are false: Pandapower Diagnostic Tool intepretation

song2208, 2021-03-26:
Hello~~~

I'm try to run the powerflow of 9 bus system in Pandapower, so The first step is create the 9 bus system in pandapower. After that I'm going to run powerflow ,but I did't complete about it. 

I use the diagnostic Tool for check an error ,so I have three noticfication  of errors  like these.

_____________ PANDAPOWER DIAGNOSTIC TOOL _____________ 

Checking for overload...

Overload check failed: Power flow still does not converge with load and generation scaled down to 0.1 percent

 --------

Checking switch configuration...

Power flow still does not converge with all switches closed.

 --------

numba_comparison failed due to the following error:
Power Flow nr did not converge after 10 iterations!

 --------

_____________ END OF PANDAPOWER DIAGNOSTIC _____________ 
{'overload': {'generation': False, 'load': False},
 'wrong_switch_configuration': False}

##############################################################################

#This is my code to create 9 bus system in pandapower. 

import pandapower as pp
import numpy as np

#create empty net
net = pp.create_empty_network()

#create buses
bus0 = pp.create_bus(net, vn_kv=16.5, max_vm_pu=1.05, min_vm_pu=0.95)
bus1 = pp.create_bus(net, vn_kv=18, max_vm_pu=1.05, min_vm_pu=0.95)
bus2 = pp.create_bus(net, vn_kv=13.8, max_vm_pu=1.05, min_vm_pu=0.95)
bus3 = pp.create_bus(net, vn_kv=220, max_vm_pu=1.05, min_vm_pu=0.95)
bus4 = pp.create_bus(net, vn_kv=220, max_vm_pu=1.05, min_vm_pu=0.95)
bus5 = pp.create_bus(net, vn_kv=220, max_vm_pu=1.05, min_vm_pu=0.95)
bus6 = pp.create_bus(net, vn_kv=220, max_vm_pu=1.05, min_vm_pu=0.95)
bus7 = pp.create_bus(net, vn_kv=220, max_vm_pu=1.05, min_vm_pu=0.95)
bus8 = pp.create_bus(net, vn_kv=220, max_vm_pu=1.05, min_vm_pu=0.95)

#create tranfo
tr1 = pp.create_transformer_from_parameters(net, bus3, bus0, 250, 220, 16.5, 14.4, 3, 0, 0 )
tr2 = pp.create_transformer_from_parameters(net, bus6, bus1, 200, 220, 18, 12.5, 3, 0, 0 )
tr3 = pp.create_transformer_from_parameters(net, bus8, bus2, 150, 220, 13.8, 8.79, 3, 0, 0 )

#create lines
l1 = pp.create_line(net, bus4, bus3, 20, "490-AL1/64-ST1A 220.0")
l2 = pp.create_line(net, bus6, bus4, 20, "490-AL1/64-ST1A 220.0")
l3 = pp.create_line(net, bus6, bus7, 20, "490-AL1/64-ST1A 220.0")
l4 = pp.create_line(net, bus7, bus8, 20, "490-AL1/64-ST1A 220.0")
l5 = pp.create_line(net, bus8, bus5, 20, "490-AL1/64-ST1A 220.0")
l6 = pp.create_line(net, bus5, bus3, 20, "490-AL1/64-ST1A 220.0")

#create gen
g0 = pp.create_gen(net, bus0, p_mw=247.5, min_p_mw=0, max_p_mw=450, vm_pu=1, slack=True, controllable=True)
g1 = pp.create_gen(net, bus1, p_mw=163.2, min_p_mw=0, max_p_mw=450, vm_pu=1, slack=True, controllable=True)
g2 = pp.create_gen(net, bus2, p_mw=163.2, min_p_mw=0, max_p_mw=450, vm_pu=1, slack=True, controllable=True)

#create loads
pp.create_load(net, bus4, p_mw=93.2, q_mvar=57.8, controllable=False)
pp.create_load(net, bus5, p_mw=66.6, q_mvar=41.3, controllable=False)
pp.create_load(net, bus7, p_mw=55.3, q_mvar=34.3, controllable=False)

net
#########################################################################

Could you tell me, What should I doing next or checking some things in my system.
Thanks you in advance!

wangzhenassd, 2021-03-30:
Hi, I think you can not set all the generators as slack (voltage reference), you can try set only the first gen as slack.

rbolgaryn, 2021-03-30:
Hi @song2208 ,

please check your transformer parameters. In your example, vk_percent is 3 and vkr_percent is 8,79-14,40. I think this is not plausible. Maybe you used one for the other?

Roman

rbolgaryn, 2021-06-03:
Closing due to no activity

closed 2021-06-03 by rbolgaryn