e2nIEE/pandapower #1187: Feature: Toggle On the contribution of distributed generators in Minimum short circuit calculation

giovannigasparini, 2021-03-30:
Hi, first of all, I would like to thank you for this awesome library! We are making extensive use of it, to the point where we would like to see additional features inside it.

Today the computation of minimum short circuit currents by **pandapower.shortcircuit.calc_sc**, case "min”,
does not consider the contribution of distributed generation, like photovoltaic power station units.
This behavior complies with the IEC 60909-0 7.1.2

Nevertheless would be interesting to toggle On this contribution, maybe by introducing an optional parameter in the method **pandapower.shortcircuit.calc_sc**, with default setting at False, to guarantee retrocompatibility.

Including this contribution to the short circuit currents would be useful to verify the coordination between all circuit breakers connected to the same busbar. For example a switch that acts as a protection, has a time of intervention that depends on the sum of the currents that flow through it. To compute the time of intervention we would need the total current flowing through the switch, therefore also the contribution of distributed generation would be relevant for this computation.

rbolgaryn, 2021-03-30:
Hi Giovanni, 

can you provide a small example of the grid or the grid you are using in this calculation?

Roman

wangzhenassd, 2021-03-30:
Hi @giovannigasparini,

Yes, that sounds good @rbolgaryn. It would be nice, if you can offer a minimum example with its "should" result and add to the short circuit test marked with xfail. We or you can then try to update the code to solve the issue.

giovannigasparini, 2021-03-31:
Hi @rbolgaryn and @wangzhenassd, thanks for the fast feedback.  We will produce a pandapower network, together with its calc_sc(case="min") output and the "shall be" output with the proposed feature. It will be a minimal network with distibutred generation. 
As soon as we have it I will post it here.

SteffenMeinecke, 2021-11-11:
Hey @giovannigasparini its some time ago, but nevertheless we would be happy to get your input.

giovannigasparini, 2021-11-11:
Hi @SteffenMeinecke,
thank you for the update. Unfortunately I don't have the chance to work into this anymore, so I am going to close this issue.

closed 2021-11-11 by giovannigasparini