e2nIEE/pandapower #1192: Question: Is there any way to relax some constraints to force opf convergence?

Crueda96, 2021-04-03:


rbolgaryn, 2021-04-15:
Hi @Crueda96 ,

I know that matpower has a feature like this, that allows substitution of constraints with costs. But in pandapower we don't have it.
@MMajidi137 is it possible when using PowerModels?

Roman

MMajidi137, 2021-04-15:
@Crueda96  Different possible relaxation are introduced in [PowerModels.jl](https://github.com/lanl-ansi/PowerModels.jl). However, the relaxation features are not available in the pandapower/develop current version. To add new features and tools related to [PowerModels.jl](https://github.com/lanl-ansi/PowerModels.jl), you need to checkout to the julia_pkg branch and follow the development structure in  [PandaModels.jl](https://github.com/e2nIEE/PandaModels.jl).

MMajidi137, 2021-04-15:
> Hi @Crueda96 ,
> 
> I know that matpower has a feature like this, that allows substitution of constraints with costs. But in pandapower we don't have it.
> @MMajidi137 is it possible when using PowerModels?
> 
> Roman

@Crueda96  Different possible relaxation are introduced in [PowerModels.jl](https://github.com/lanl-ansi/PowerModels.jl). However, the relaxation features are not available in the pandapower/develop current version. To add new features and tools related to [PowerModels.jl](https://github.com/lanl-ansi/PowerModels.jl), you need to checkout to the julia_pkg branch and follow the development structure in  [PandaModels.jl](https://github.com/e2nIEE/PandaModels.jl).

MMajidi137, 2021-06-14:
@Crueda96 Did you solve the issue? you also can easily set the upper and lower bounds of the constraints that you want to relax in very big value( not exactly infinity).

closed 2021-06-24 by rbolgaryn