e2nIEE/pandapower #1197: Potentially depreciated name for the asymmetric_sgen used + why division by 1000 (old to new convention?)

loshmee, 2021-04-07:
sgen_3ph produces an error in execution. When replaced by asymmetric_sgen, the error disappears

[image]

lthurner, 2021-04-07:
From what I understand, this code is never executed, because the mode has been switched to pf_3ph. So I think this code should be deleted completely.

The load data is currently added in this function:

https://github.com/e2nIEE/pandapower/blob/1b85987696e67841abc00bd48e0e5ebf4f191bb1/pandapower/pf/runpp_3ph.py#L128-L152

You said you are getting an error? Can your reproduce that with a clean installation or did you maybe change something in the if condition as well that this code is executed?

lthurner, 2021-04-07:
OK I just figured out what you mean. This code is actually used in symmetric power flow, not in asymmetric. The asymmetric loads are summed up here for the balanced power flow. 

This needs to be fixed and a test needs to be added that runs a balanced power flow with the asymmetric loads and sgens. The division by 1000 is also incorrect, as you pointed out.

lthurner, 2021-04-07:
Fixed and merged, symmetric power flow should now be working properly with asymmetric loads and sgens. Thanks for reporting!

closed 2021-04-07 by lthurner