e2nIEE/pandapower #1200: Question: Is a constant minimum loading percent of MV-LV transformer of CIGRE LV network strange?

Pimpussorn, 2021-04-08:
Hi everyone,

I ran time series simulations with create_cigre_network_lv() and run_timeseries() and when I logged my MV-LV transformer data, I always got the minimum loading percent at around 26 no matter how much I changed the load input. I didn't change any default parameter of the grid; I only imported load time series to nodes I chose.  Does anyone know where to check this problem? It seems like the loading percent is cut and not allowed to go below a specific limit which I don't know.

Thanks in advance.

wangzhenassd, 2021-04-10:
Hi,
 can you share your code to reproduce this?

Pimpussorn, 2021-04-16:
@wangzhenassd 
Hi, 
I have push my code on my public repository https://github.com/Pimpussorn/Test_CIGRE_LV_net you can check it out.

wangzhenassd, 2021-04-17:
Hi,
by adapting the create_controller with 
```
def create_controllers(net, ds):
    for load_ix in net.load.index:
        ConstControl(net, element='load', variable='p_mw', element_index=[load_ix],
                      data_source=ds, profile_name=["load1_p"])
```
The result seems to be correct. The change of load p has impact on the trafo loading. In your original code, it seems to only update a small number of loads in the network.
[image]

wangzhenassd, 2021-04-20:
@Pimpussorn Is this question solved?

Pimpussorn, 2021-04-20:
@wangzhenassd thank you so much for the reply. It is really helpful. I still have the question left that the minimum value of the first column (transformer with the index 0) seems to vary only at 28 like your results showing. When I plotted the results, there are times when the values seem to be cut at this value (28). So I'm wondering whether I have to change the parameters of this transformer or there is anyway to trace back how the current at the low and high voltage side of the transformer is calculated for the loading percent according to this grid, since my input was power. Or if you have any suggestion, I would really appreciate it.
[image]

wangzhenassd, 2021-05-22:
Hi @Pimpussorn , sorry for the delayed answer, I finally find the time to dig into your problem. The lower point is caused the q_mvar of the load, which is considered as constant (default value in the pandapower network) in the timeseries simulation. If you set it manually to 0, then the problem will disappear.

Pimpussorn, 2021-05-30:
@wangzhenassd Thank you, so much. Do you know how I can check the default value q_mvar of the load?

wangzhenassd, 2021-05-30:
The q_mvar in pandapower networks will be used by default. It can be checked with net.load.q_mvar

Pimpussorn, 2021-05-30:
@wangzhenassd I really can't thank you enough. Thank you so much for your time, you just saved my life. I wish you a nice weekend.

wangzhenassd, 2021-05-30:
You are very welcome : ) Can this issue be closed?

Pimpussorn, 2021-05-31:
@wangzhenassd yes, it can be closed now.

closed 2021-05-31 by wangzhenassd