e2nIEE/pandapower #1202: Calculating pandapower susceptances

leonlan, 2021-04-09:
For calculating DCPF equations I'm trying to obtain the line susceptances of some IEEE test cases. I assume DC approximation (so the susceptances are approximated as the inverse of the reactance) and calculate it using two ways:

1. Directly using the `pandadower` line and transformer parameters
2. Digging into the `pypower` library and obtaining the susceptances parameters

See this [gist](https://gist.github.com/leonlan/e4fcfbbdf3765d1cf8c260b08371ab1f) for the code and a few examples of test cases. Here is the problem I don't understand:
- Why are some pandapower line susceptances very small (e.g. << 1) whereas others are much larger (in the order of 10^1-10^3)?
- Why do some pandapower line susceptances differ by several orders of magnitudes w.r.t. the pypower susceptances?

wangzhenassd, 2021-04-10:
Hi,
I tried your code, I think to calculate b for lines in p.u. you also need the vn_kv of the line, which is missing in your equation.

`b_lines = np.array(1/(net.line['x_ohm_per_km']*net.line['length_km']*net.sn_mva/net.line['parallel'])) * net.bus.loc[net.line.from_bus.values, "vn_kv"].values ** 2
`
`
array([1690.04563123,  448.35007174, 1183.43195266,  369.84984096,
        520.64351539,  500.30018011,  287.33980806,  505.12703945,
        567.15063521,  575.10927076,  584.69274396, 2374.73284256,
        502.76520865,  390.91513232,  767.63644738])`

leonlan, 2021-04-12:
> Hi,
> I tried your code, I think to calculate b for lines in p.u. you also need the vn_kv of the line, which is missing in your equation.
>
> `b_lines = [...]

Thanks for the help!

closed 2021-04-12 by leonlan