e2nIEE/pandapower #1206: WARNING: replacing module PP2PM with PowerModel.jl usage

Crueda96, 2021-04-14:
There is a way to avoid WARNING: replacing module PP2PM? I'm using PowerModels.jl for solving an OPF AC. The solution converges however each time I execute the function the warning is displayed

I'm using spyder ide and anaconda prompt and in both warnings are displayed

MMajidi137, 2021-04-15:
@Crueda96 to avoid the warning, you need to checkout to the julia_pkg branch and follow the structure in [PandaModels.jl](https://github.com/e2nIEE/PandaModels.jl).

SteffenMeinecke, 2021-11-11:
issue is outdated due to several merges to develop

closed 2021-11-11 by SteffenMeinecke

angelinasyrri, 2022-03-16:
Hi! I also face the exact same issue as above. I am using Julia 1.6.5 with windows and  Spyder through Anaconda. I can't understand from your replies above how to resolve the issue. Thanks!

MMajidi137, 2022-03-16:
you may use Julia 1.5 and not use anaconda, then pull the develop branch, the develop branch uses the PandaModels package instead of the pp2pm module. @angelinasyrri

angelinasyrri, 2022-03-16:
Ηi! Thanks for the reply:) I don't really get it how 'the develop branch uses the PandaModels package instead of the pp2pm module'. I have pulled the develop branch of PandaModels.jl, and saved it on the directory that I work, but I don't expect this to affect somehow which module will be used when I am running for instance 'pp.runpm_ac_opf(net, verbose=True)'. I guess what affects everything is what I import in python before I start , which is what you have suggested in the installation process of 'Optimization with PandaModels.jl'. I have imported 'from julia.PowerModels import run_ac_opf'. How can I somehow import PandaModels package? Can you provide more exact steps? Shall I use a different command than pp.runpm_ac_opf?

angelinasyrri, 2022-03-18:
Hi @MMajidi137! Everything works now great! many thanks for the support!