e2nIEE/pandapower #1207: Data is wrong in converted ppc network

lvzhibai, 2021-04-16:
I found a bug in pandapower about converted ppc net.
For any pandapower networks with more than 10 bus/line/gen, `net[element]` is not sorted.
```python
>>>import pandapower as pp
>>>import pandapower.networks as pn
>>>net = pn.case14()   # take case 14 as example
>>>net["bus"].index 
Int64Index([0, 1, 10, 11, 12, 13, 2, 3, 4, 5, 6, 7, 8, 9], dtype='int64')  # Not sorted
```

When I do

```python
pp.runpp(net)
net["_ppc"] = pp.converter.to_ppc(net)
```
or any other similar commands which converts net to ppc format, the output data is wrong.

In ppc data, all elements index are sorted, e.g. `net["_ppc"]["bus"][:, BUS_I] = array([0, 1, 2, ..., 13])`.

And that causes wrong output. There should be codes about sorting, or warning/error message during processing, but they are missing.

However, the power flow and OPF results are correct. 

So I suggest either correct each ppc result `net["_ppc"][element]` or sort elements before executing `to_ppc`.

Actually, without `pp.runpp(net)`, there is an error message when running `pp.converter.to_ppc(net)` about mismatch between `net[element].index` and `net[res_element].index`, but it is another logic.

lvzhibai, 2021-04-16:
@SteffenMeinecke

SteffenMeinecke, 2021-04-19:
@lvzhibai Thank you for your contribution!
As far as I evaluate the issue, everything (the converter from_ppc/to_ppc) is correct. The only _problem_ is that when reading json files, the tables are not sorted by the index.
It can be easily fixed by
```
for elm in pp.pp_elements():
    net[elm].sort_index(inplace=True)
```

Since also other users already stumbled upon this topic, I suggest to sort the tables in `io_file.py`.
@ascheidl If `from_json_dict()` is used, it would be easy to add a new line `net[key] = net[key].sort_index()` after line 319 of `io_file.py`. But whit if `PPJSONDecoder()` is used? Is there also an elegant/pythonic way of coding?

SteffenMeinecke, 2021-04-19:
I noticed that with the current version of to_json() and from_json() the sorting is maintained and not lost. So the best solution from my point of view would be to simply update all json files (including sorting).

commit fa9ea48c (2021-04-20): close #1207
closed 2021-04-20 by SteffenMeinecke