e2nIEE/pandapower #1209: to_mpc not working for some of the pn.networks

BDonnot, 2021-04-20:
Hello,

I wanted to report an issue when trying to convert grids in "pandapower.networks" to "mpc" format.

I am using pandapower version `2.6.0` on ubuntu linux 20.04  with python 3.8.5

Here is the code i did (taken directly from the example):

```python
import pandapower as pp
import pandapower.networks as pn
net = pn.case9()
pp.converter.to_mpc(net=net, filename="case9.mat")
```

And the result i got:

```
---------------------------------------------------------------------------
UserWarning                               Traceback (most recent call last)
<ipython-input-1-b03f8c807a77> in <module>
      8 import pandapower.networks as pn
      9 net=pn.case9()
---> 10 pp.converter.to_mpc(net=net, filename="case9.mat")
     11 
     12 path_matpower = "."

~/.local/lib/python3.8/site-packages/pandapower/converter/matpower/to_mpc.py in to_mpc(net, filename, **kwargs)
     41 
     42     """
---> 43     ppc = to_ppc(net, **kwargs)
     44 
     45     mpc = dict()

~/.local/lib/python3.8/site-packages/pandapower/converter/pypower/to_ppc.py in to_ppc(net, calculate_voltage_angles, trafo_model, switch_rx_ratio, check_connectivity, voltage_depend_loads, init, mode)
    100                      recycle=None, voltage_depend_loads=voltage_depend_loads)
    101     #  do the conversion
--> 102     _, ppci = _pd2ppc(net)
    103     ppci['branch'] = ppci['branch'].real
    104 #    ppci.pop('internal')

~/.local/lib/python3.8/site-packages/pandapower/pd2ppc.py in _pd2ppc(net, sequence)
     70 
     71     # generate ppc['bus'] and the bus lookup
---> 72     _build_bus_ppc(net, ppc)
     73     if sequence == 0:
     74         from pandapower.pd2ppc_zero import _add_ext_grid_sc_impedance_zero, _build_branch_ppc_zero

~/.local/lib/python3.8/site-packages/pandapower/build_bus.py in _build_bus_ppc(net, ppc)
    303     if mode != "nx":
    304         set_reference_buses(net, ppc, bus_lookup, mode)
--> 305     vm_pu = get_voltage_init_vector(net, init_vm_pu, "magnitude")
    306     if vm_pu is not None:
    307         ppc["bus"][:n_bus, VM] = vm_pu

~/.local/lib/python3.8/site-packages/pandapower/build_bus.py in get_voltage_init_vector(net, init_v, mode)
    212                 # cannot init from results, since sorting of results is different from element table
    213                 # TO BE REVIEWED! Why there was no raise before this commit?
--> 214                 raise UserWarning("Init from results not possible. Index of res_bus do not match with bus. "
    215                             "You should sort res_bus before calling runpp.")
    216                 return None

UserWarning: Init from results not possible. Index of res_bus do not match with bus. You should sort res_bus before calling runpp.
```

The same error occur with the case118:
```python
import pandapower as pp
import pandapower.networks as pn
net=pn.case118()
pp.converter.to_mpc(net=net, filename="case118.mat")
```

Also, i wanted to point out that the documentation of the to_mpc (https://pandapower.readthedocs.io/en/v2.0.1/converter/matpower.html#pandapower.converter.to_mpc) has an issue with the layout:
- the "example" is not displayed properly within a python-like shell (probably missing something like `.. code-block:: python`)
- the `optional`  section is also buggy, in the sense that it emphasizes the `filename (str, None) - File path + name of the mat file which will be created. If None`. I believe it's an issue related to the `-` preceeding the sentence "- File path" that is wrongly interpreted as marking the beginning of a section.

BDonnot, 2021-04-20:
I managed to fix the issue with the following code:
```python
import pandapower as pp
import pandapower.networks as pn
net=pn.case118()
pp.runpp(net)  # added 
pp.converter.to_mpc(net=net, filename="case118.mat")
```

The trick is then to call a powerflow before converting it to matpower.

This should, in my opinion, either be specified explicitly in the documentation, or be made directly in the "to_mpc" function.

rbolgaryn, 2021-04-22:
Hi @BDonnot ,

thank you for the comment, I think we should include it in the documentation.

Best regards, Roman

commit ec5ee76a (2021-11-16): fix init behaviour and add error messages. fix #1209
closed 2021-11-16 by SteffenMeinecke