e2nIEE/pandapower #1210: Robust LAV estimator is so slow

afeser, 2021-04-20:
Hi,

When I try to solve LAV state estimation with 600+ nodes, I get stuck when linprog of scipy waits too long to converge. Therefore, I integrated Google OR tools to the repo by changing the `estimation/algorithm/lp.py` file. It now converges much faster than the original version. The system that did not converge in an hour using scipy linprog has converged in a minute when I used OR tools.

I want to contribute to the repository, and I had a look into the `contributing` file. However, I wanted to be sure whether it is okay to integrate or-tools to the repository directly. Are there other problems such as licence issues or being reluctant to add new dependencies?

wangzhenassd, 2021-04-20:
Hi @afeser, thanks a lot for your contribution! It would be a very nice enhancement to the current state estimation tool! Please feel free to open a PR!
However, I think the OR tools might require CPP Compiler for users on windows, which is probably not trivial to configure correctly. Thus, I would suggest that it should not be added to the dependencies. The functionality itself could be added to the pandapower code with extra check if the OR tools is available. 

To make this work, an example you can find would be with the external loadflow solver named lightsim2grid. If this module is installed, this function is then available for the users. The test should be marked with the extra availability check.
https://github.com/e2nIEE/pandapower/blob/1462ef06906a98cd7bad20b2e98b7e475a1dc6c7/pandapower/pf/run_newton_raphson_pf.py#L28-L33
https://github.com/e2nIEE/pandapower/blob/1462ef06906a98cd7bad20b2e98b7e475a1dc6c7/pandapower/test/loadflow/test_runpp.py#L1238-L1247

afeser, 2021-04-20:
Ok, it seems conditional library loading is very reasonable, thank you!

I will implement it in my free time with a similar approach you provided.

afeser, 2021-04-22:
@wangzhenassd I implemented the OR-tools, and a simple tester adding 2 methods to  `test_opt_lp_estimation.py`. However, I am not sure if I did it correctly. Can you check it?

I tested, and the test result is:
```
556 passed, 29 skipped, 13 xfailed, 1 xpassed, 122 warnings in 361.66s (0:06:01)
```
Before any changes by me, it was:
```
554 passed, 29 skipped, 13 xfailed, 1 xpassed, 122 warnings in 361.66s (0:06:01)
```

If everything is fine, I will create a pull request.

The commits are: [Commit 1](https://github.com/afeser/pandapower/commit/a27bb1bfbe64168e5fe8c0d039f6c8bc8052cb58) [Commit 2](https://github.com/afeser/pandapower/commit/e072700fe51f9fd5bcfd33122d45661dd1285c22)

wangzhenassd, 2021-04-22:
Hi @afeser, it looks pretty good to me. Please open a pull request. There we can see if the tests still work as expected and whether further adaptions are required.  Thanks a lot again for your contribution!

afeser, 2021-04-24:
I guess there is no need for this issue after the pull request is merged.

Thank you @wangzhenassd for your help! I am closing the issue.

closed 2021-04-24 by afeser

wangzhenassd, 2021-04-25:
@afeser You are very welcome! Thanks again for your contribution! I am now updating the github actions so that ortools can be included in the test, it seems to work now.