e2nIEE/pandapower #1217: pandapower not supported in my PyCall

lvzhibai, 2021-04-25:
I am trying to using pandapower in julia with PyCall.jl.
When I do
```julia
julia> using PyCall

julia> pyimport("math")
PyObject <module 'math' from '/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/lib-dynload/math.cpython-39-darwin.so'>

julia> pyimport("os")
PyObject <module 'os' from '/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/os.py'>

julia> pyimport("matplotlib")
PyObject <module 'matplotlib' from '/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/matplotlib/__init__.py'>

julia> pyimport("scipy")
PyObject <module 'scipy' from '/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/scipy/__init__.py'>
```
They are imported normally.
But when I try
```julia
julia> pyimport("pandapower")
```
It has error and returns
```julia
Assertion failed: (findOption(Name) == Values.size() && "Option already exists!"), function addLiteralOption, file /Users/ci/miniconda3/conda-bld/llvmdev_1601977477900/work/include/llvm/Support/CommandLine.h, line 853.

signal (6): Abort trap: 6
in expression starting at none:1
__pthread_kill at /usr/lib/system/libsystem_kernel.dylib (unknown line)
Allocations: 33012935 (Pool: 33004496; Big: 8439); GC: 29
```

And Julia crashed.
Same case by simbench.

In error message `Assertion fail`, the `ci` folder is not even exist in `/Users`, and I don't have miniconda.

System: MacOS Big Sur
Python: 3.9.3 standalone
Julia: 1.5.4

MMajidi137, 2021-04-29:
@lvzhibai did you try:
`using PyCall; py"""
your python code"""
`

lvzhibai, 2021-04-29:
It's the same. pandapower cannot be imported.
```julia
julia> py"""
       import numpy as np

       def sinpi(x):
           return np.sin(np.pi * x)
       """

julia> py"""
       import os
       import matplotlib
       """

julia> py"""
       import pandapower
       """
Assertion failed: (findOption(Name) == Values.size() && "Option already exists!"), function addLiteralOption, file /Users/ci/miniconda3/conda-bld/llvmdev_1601977477900/work/include/llvm/Support/CommandLine.h, line 853.

signal (6): Abort trap: 6
in expression starting at none:1
__pthread_kill at /usr/lib/system/libsystem_kernel.dylib (unknown line)
Allocations: 39836026 (Pool: 39825359; Big: 10667); GC: 36

Julia has exited.
```

ssehztirom, 2021-05-02:
Have you been able to resolve the issue? I am experiencing the same problem on a Mac when trying to import [scvi](https://docs.scvi-tools.org/en/stable/) via PyCall.

System
_______
ProductName:	macOS
ProductVersion:	11.3
BuildVersion:	20E232
_____
Julia  Version 1.5.3 (2020-11-09)
____
  "Conda"  => v"1.5.2"
  "IJulia" => v"1.23.2"
  "PyCall" => v"1.92.3"

MMajidi137, 2021-05-03:
@ssehztirom what error did you get?

ssehztirom, 2021-05-03:
Hi @MMajidi137,

I get the same error as mentioned above:

pyimport("scvi")
Assertion failed: (findOption(Name) == Values.size() && "Option already exists!"), function addLiteralOption, file /Users/ci/miniconda3/conda-bld/llvmdev_1601977477900/work/include/llvm/Support/CommandLine.h, line 853.

signal (6): Abort trap: 6
in expression starting at REPL[5]:1
__pthread_kill at /usr/lib/system/libsystem_kernel.dylib (unknown line)
Allocations: 11878922 (Pool: 11874927; Big: 3995); GC: 13

It seems to be Mac specific (don't get it on a Linux server) and it appears from a certain version of scvi onwards (0.6.8+; https://pypi.org/project/scvi/).

MMajidi137, 2021-05-03:
> Hi @MMajidi137,
>
> I get the same error as mentioned above:
>
> pyimport("scvi")
> Assertion failed: (findOption(Name) == Values.size() && "Option already exists!"), function addLiteralOption, file [...]

There is an open issue in pycall repo, issue [#861](https://github.com/JuliaPy/PyCall.jl/issues/861). It may be LLVM issue, u may remove pycall and set path again. follow the step from PyCall [repo](https://github.com/JuliaPy/PyCall.jl)

ssehztirom, 2021-05-04:
Thanks! I will try that.

Best

Moritz
Moritz Hess
Institute of Medical Biometry and Statistics
Faculty of Medicine and Medical Center - University of Freiburg
01.009
Stefan-Meier-Straße 26
D-79104 Freiburg
Phone.: +49 761 203 6670

ppalmes, 2021-05-20:
I have the same issue:
aif360 = pyimport("lale.lib.aif360")
Assertion failed: (findOption(Name) == Values.size() && "Option already exists!"), function addLiteralOption, file /Users/ci/miniconda3/conda-bld/llvmdev_1601977477900/work/include/llvm/Support/CommandLine.h, line 853.

signal (6): Abort trap: 6
in expression starting at REPL[5]:1
__pthread_kill at /usr/lib/system/libsystem_kernel.dylib (unknown line)
Allocations: 10405176 (Pool: 10401439; Big: 3737); GC: 14
[1]    9963 abort      julia --project

vogt31337, 2025-03-31:
I also had this issue, the solution is quite awful: Julia uses a specific version of libllvm and pandapower uses numba which also internally uses libllvm. So these two versions need to use the same library. You have to up-/downgrade julia and numba accordingly, otherwise it will just crash with a segfault, since the os is not able to load different versions of the same library in the same context.

closed 2025-03-31 by vogt31337