e2nIEE/pandapower #1223: res_cost Not Matching Manual Objective Calculation for Some Test Cases

kravitsjacob, 2021-05-06:
I wanted to preface my issue with the fact that I love this tool. It has greatly improved my research workflow. My issue is with the optimal power flow functionality:

I want to manually verify the value of the optimal cost that is returned via `net.res_cost`. I provided a simple code below to extract the active power outputs and do the objective function calculation via pandas column operations. The code works for some cases and not others. Does anyone know why these values don't match for some cases? Is this an issue with `net.res_cost`? Thank you in advance!

```
import pandapower.networks
import pandapower
import pandas as pd

def getCostObjective(net):
    df_gen = pd.DataFrame()
    df_ext_grid = pd.DataFrame()
    # Generators
    df_gen['PP Network Index'] = net.res_gen.index.to_list()
    df_gen['PP Bus Type'] = 'gen'
    df_gen['p_mw'] = net.res_gen['p_mw']
    # External Grid
    df_ext_grid['PP Network Index'] = net.res_ext_grid.index.to_list()
    df_ext_grid['PP Bus Type'] = 'ext_grid'
    df_ext_grid['p_mw'] = net.res_ext_grid['p_mw']
    # Combine
    df_p = pd.concat([df_gen, df_ext_grid])
    df_p = df_p.reset_index()
    cost_df = net.poly_cost.merge(df_p, left_on=['element', 'et'], right_on=['PP Network Index', 'PP Bus Type'])
    # Manually Computing Cost Objective
    cost_obj = (cost_df['cp0_eur'] + cost_df['cp1_eur_per_mw'] * cost_df['p_mw'] + cost_df['cp2_eur_per_mw2'] * cost_df[
        'p_mw'] ** 2).sum()
    return cost_obj

# Example with 30 Case
net = pandapower.networks.case30()
pandapower.runopp(net)
print(net.res_cost == getCostObjective(net))
# Example with 118 Case
net = pandapower.networks.case118()
pandapower.runopp(net)
print(net.res_cost == getCostObjective(net)) 
```

Note: The above code is fairly hardcoded and not meant to be a robust method of doing the manual objective calculation. It is simply meant to illustrate my issue.

wangzhenassd, 2021-05-06:
Hi, can you offer an example, in which your example does not work? In case30 and case118 your code seems to work.

kravitsjacob, 2021-05-06:
Thank you for your response! The above code did yield same results for case30 but did _not_ for case118. I am using python 3.7.10 with pandapower 2.6.0. I have attached a screenshot below:

[image]

Note, I put the above code in the file test.py. 

Could you tell me which version you got the test to work successfully so I can try on my machine?

wangzhenassd, 2021-05-07:
I am using the numpy 1.19.4, pandas 1.2.0, scipy 1.3.3.
Can you maybe show the difference between the two costs? Maybe it is an accuracy issue of floating number.

wangzhenassd, 2021-05-10:
```
case30
Pandapower Cost:  578.4859431403698
Cost from my function:  578.4859431403698
case118
Pandapower Cost:  129695.1018738289
Cost from my function:  129695.10187382888
```
This issue is strange. It is correct from my test.

kravitsjacob, 2021-05-21:
I agree that this is a strange issue. Just so you can reproduce the issue, [here](https://github.com/kravitsjacob/pandapower_rescost) is a link to a very simple repository with a dockerfile. 

Here is a screenshot of my run of the test in the attached repository:
[image]

SteffenMeinecke, 2021-11-11:
@kravitsjacob Does this issue still exist for you?
Seems like both of you @kravitsjacob @wangzhenassd couldn't solve it.
Since this is an hard to reproduce corner case, I close this issue in hope that it doesn't occur again. If you or others have this problem again, please reopen.

Just a remark: net.res_cost ist available with 2eb29ea.

closed 2021-11-11 by SteffenMeinecke

kravitsjacob, 2021-11-11:
Sounds good!