e2nIEE/pandapower #1228: Problems with 3 phase asymmetric power flow

nuriapastorcal, 2021-05-12:
Hello everyone,

I am Nuria Pastor,  I’m a student at Polytechnic University of Valencia. I am doing my end of degree project with pandapower. I am writing this post to see if someone can help me with an error that is appearing to me when I try to run a 3 phase asymmestric power flow. I leave you the code right under this message and the error that pops up to see is you can help me understand what is wrong with the power flow.

Thank you so much in advanced.

Nuria.

The code:

import numpy as np
import pandapower as pp
import pandas as pd

from pandapower.pf.runpp_3ph import runpp_3ph

net=pp.create_empty_network()

b1=pp.create_bus(net, 0.4) #E1
b2=pp.create_bus(net, 0.4) #EmbarradoA
b3=pp.create_bus(net, 0.4) #EmbarradoB
b4=pp.create_bus(net, 0.4) #EmbarradoC
b5=pp.create_bus(net, 20.0) #RedMT
b6=pp.create_bus(net, 0.4) #Terminal
b7=pp.create_bus(net, 0.4) #Terminal1
b8=pp.create_bus(net, 0.4) #Terminal10
b9=pp.create_bus(net, 0.4) #Terminal11
b10=pp.create_bus(net, 0.4) #Terminal12
b11=pp.create_bus(net, 0.4) #Terminal13
b12=pp.create_bus(net, 0.4) #Terminal14
b13=pp.create_bus(net, 0.4) #Terminal15
b14=pp.create_bus(net, 0.4) #Terminal16
b15=pp.create_bus(net, 0.4) #Terminal17
b16=pp.create_bus(net, 0.4) #Terminal18
b17=pp.create_bus(net, 0.4) #Terminal19
b18=pp.create_bus(net, 0.4) #Terminal2
b19=pp.create_bus(net, 0.4) #Terminal20
b20=pp.create_bus(net, 0.4) #Terminal21
b21=pp.create_bus(net, 0.4) #Terminal22
b22=pp.create_bus(net, 0.4) #Terminal23
b24=pp.create_bus(net, 0.4) #Terminal25
b26=pp.create_bus(net, 0.4) #Terminal27
b28=pp.create_bus(net, 0.4) #Terminal29
b29=pp.create_bus(net, 0.4) #Terminal3

pp.create_ext_grid(net, bus=b5)

pp.create_load(net, b28, p_mw=2, q_mvar=0.0, type='wye')

pp.create_asymmetric_load(net, b28, p_a_mw=0.0, p_b_mw=0.0, p_c_mw=2, q_a_mvar=0.0, q_b_mvar=0.0, q_c_mvar=0.0)

pp.create_gen(net, b4, p_mw=0.5, name='PV1')
pp.create_gen(net, b4, p_mw=0.5, name='PV2')
pp.create_gen(net, b5, p_mw=0.5, name='Eólica')

pp.create_asymmetric_sgen(net, b4, name='PV1', p_a_mw=0.5, p_b_mw=0, p_c_mw=0, q_a_mvar=0, q_b_mvar=0, q_c_mvar=0)
pp.create_asymmetric_sgen(net, b4, name='PV2', p_a_mw=0, p_b_mw=0.5, p_c_mw=0, q_a_mvar=0, q_b_mvar=0, q_c_mvar=0)
pp.create_asymmetric_sgen(net, b5, name='Eólica', p_a_mw=0, p_b_mw=0, p_c_mw=0.5, q_a_mvar=0, q_b_mvar=0, q_c_mvar=0)

pp.create_transformer_from_parameters(net, hv_bus=b5, lv_bus=b18, sn_mva=0.25, vn_hv_kv=20.0, vn_lv_kv=0.4, vkr_percent=4.2, vk_percent=4.2, pfe_kw=0, i0_percent=0.1, vk0_percent=4.2, vkr0_percent=4.2, si0_hv_partial=0.1)
pp.create_transformer_from_parameters(net, hv_bus=b5, lv_bus=b6, sn_mva=0.25, vn_hv_kv=20.0, vn_lv_kv=0.4, vkr_percent=4.2, vk_percent=4.2, pfe_kw=0, i0_percent=0.1, vk0_percent=4.2, vkr0_percent=4.2, si0_hv_partial=0.1)

pp.create_line_from_parameters(net, from_bus=b1, to_bus=b17, length_km=1, r_ohm_per_km=1.0E-6, x_ohm_per_km=1.0E-4, c_nf_per_km=0, max_i_ka=100.0, r0_ohm_per_km=1.0E-6,x0_ohm_per_km=1.0E-4,c0_nf_per_km=0)
pp.create_line_from_parameters(net, from_bus=b1, to_bus=b29, length_km=1, r_ohm_per_km=1.0E-6, x_ohm_per_km=1.0E-4, c_nf_per_km=0, max_i_ka=100.0, r0_ohm_per_km=1.0E-6,x0_ohm_per_km=1.0E-4,c0_nf_per_km=0)
pp.create_line_from_parameters(net, from_bus=b2, to_bus=b6, length_km=1, r_ohm_per_km=1.0E-6, x_ohm_per_km=1.0E-4, c_nf_per_km=0, max_i_ka=100.0, r0_ohm_per_km=1.0E-6,x0_ohm_per_km=1.0E-4,c0_nf_per_km=0)
pp.create_line_from_parameters(net, from_bus=b2, to_bus=b21, length_km=1, r_ohm_per_km=1.0E-6, x_ohm_per_km=1.0E-4, c_nf_per_km=0, max_i_ka=100.0, r0_ohm_per_km=1.0E-6,x0_ohm_per_km=1.0E-4,c0_nf_per_km=0)
pp.create_line_from_parameters(net, from_bus=b2, to_bus=b22, length_km=1, r_ohm_per_km=1.0E-6, x_ohm_per_km=1.0E-4, c_nf_per_km=0, max_i_ka=100.0, r0_ohm_per_km=1.0E-6,x0_ohm_per_km=1.0E-4,c0_nf_per_km=0)
pp.create_line_from_parameters(net, from_bus=b2, to_bus=b24, length_km=1, r_ohm_per_km=1.0E-6, x_ohm_per_km=1.0E-4, c_nf_per_km=0, max_i_ka=100.0, r0_ohm_per_km=1.0E-6,x0_ohm_per_km=1.0E-4,c0_nf_per_km=0)
pp.create_line_from_parameters(net, from_bus=b3, to_bus=b17, length_km=1, r_ohm_per_km=1.0E-6, x_ohm_per_km=1.0E-4, c_nf_per_km=0, max_i_ka=100.0, r0_ohm_per_km=1.0E-6,x0_ohm_per_km=1.0E-4,c0_nf_per_km=0)
pp.create_line_from_parameters(net, from_bus=b3, to_bus=b21, length_km=1, r_ohm_per_km=1.0E-6, x_ohm_per_km=1.0E-4, c_nf_per_km=0, max_i_ka=100.0, r0_ohm_per_km=1.0E-6,x0_ohm_per_km=1.0E-4,c0_nf_per_km=0)
[... 95 lines cut ...]
     44     _get_bus_results_3ph(net, bus_pq)
     45 

C:\ProgramData\Anaconda3\lib\site-packages\pandapower\results_gen.py in _get_gen_results_3ph(net, ppc0, ppc1, ppc2, bus_lookup_aranged, pq_bus)
     57     # get results for gens
     58     if gen_end > eg_end:
---> 59         b, pA, qA, pB, qB, pC, qC = _get_pp_gen_results_3ph(net, ppc0, ppc1, ppc2, b, pA, qA, pB, qB, pC, qC)
     60 
     61     if not ac:

C:\ProgramData\Anaconda3\lib\site-packages\pandapower\results_gen.py in _get_pp_gen_results_3ph(net, ppc0, ppc1, ppc2, b, pA, qA, pB, qB, pC, qC)
    292 
    293 def _get_pp_gen_results_3ph(net, ppc0, ppc1, ppc2, b, pA, qA, pB, qB, pC, qC):
--> 294     pA_gen, qA_gen, pB_gen, qB_gen, pC_gen, qC_gen = _get_p_q_gen_results_3ph(net, ppc0, ppc1, ppc2)
    295     _get_v_gen_results_3ph(net, ppc0, ppc1, ppc2)
    296 

C:\ProgramData\Anaconda3\lib\site-packages\pandapower\results_gen.py in _get_p_q_gen_results_3ph(net, ppc0, ppc1, ppc2)
    204     qA, qB, qC = map(lambda x: x.flatten(), np.imag(Sabc))
    205 
--> 206     net["res_gen_3ph"]["p_a_mw"] = pA
    207     net["res_gen_3ph"]["p_b_mw"] = pB
    208     net["res_gen_3ph"]["p_c_mw"] = pC

KeyError: 'res_gen_3ph'

rbolgaryn, 2021-05-21:
Hi @nuriapastorcal ,

Can you narrow the error down with a more minimalistic example?

Roman

SteffenMeinecke, 2021-11-11:
Close this issue due to inactivity.

pljavals, 2023-01-05:
I had the same problem when I tried to run a 3 phase power flow, if someone could help me, this is my code:

import pandapower
import pandapower.networks
import pandapower.topology
import pandapower.plotting
import pandapower.converter
import pandapower.estimation
import pandapower as pp

#NETWORK
#create an empty network
net = pp.create_empty_network()

#BUSES
#We now create the three high voltage (vn_kv=110.) and four medium voltage (vn_kv=20.) buses.
bus1 = pp.create_bus(net, name="Birch", vn_kv=138)
bus2 = pp.create_bus(net, name="Elm", vn_kv=138)
bus3 = pp.create_bus(net, name="Pine", vn_kv=138)
bus4 = pp.create_bus(net, name="Maple", vn_kv=138)

#EXTERNAL GRID
#We now create an external grid connection that serves as slack node for the power flow calculation. The voltage of the external grid is set to a magnitude of 1.02 per unit and 50 degrees voltage angle:
pp.create_ext_grid(net, bus1, vm_pu=1.00, va_degree=0, s_sc_max_mva=1000, rx_max=0.1, x0x_max=1.0, r0x0_max=0.1) # Create an external grid connection
net.ext_grid 
#show external grid table

#LINES
#The network includes three medium voltage lines and one high voltage line. The bus connections and line lengths are defined in the network diagram:
#The line parameters are once again taken from the standard type library (see standard type library tutorial). We use different line lengths and standard types for each line:
line1 = pp.create_line_from_parameters(net, bus1, bus2, length_km=1, r_ohm_per_km=1.777, x_ohm_per_km=8.887, c_nf_per_km=0, g_us_per_km=0.002, g0_us_per_km=0.002, r0_ohm_per_km=1.777, x0_ohm_per_km=8.887, c0_nf_per_km=0, max_i_ka=None,  name="Line 1")
line2 = pp.create_line_from_parameters(net, bus1, bus3, length_km=1, r_ohm_per_km=1.312, x_ohm_per_km=6.56, c_nf_per_km=0, g_us_per_km=-0.013, g0_us_per_km=-0.013, r0_ohm_per_km=1.312, x0_ohm_per_km=6.56, c0_nf_per_km=0, max_i_ka=None, name="Line 2")
line3 = pp.create_line_from_parameters(net, bus2, bus4, length_km=1, r_ohm_per_km=1.312, x_ohm_per_km=6.56, c_nf_per_km=0, g_us_per_km=-0.013, g0_us_per_km=-0.013, r0_ohm_per_km=1.312, x0_ohm_per_km=6.56, c0_nf_per_km=0, max_i_ka=None, name="Line 3")
line4 = pp.create_line_from_parameters(net, bus3, bus4, length_km=1, r_ohm_per_km=2.243, x_ohm_per_km=11.215, c_nf_per_km=0, g_us_per_km=0.002, g0_us_per_km=0.002, r0_ohm_per_km=2.243, x0_ohm_per_km=11.215, c0_nf_per_km=0, max_i_ka=None, name="Line 4")
#The full line table looks like this:
net.line # show line table

#LOAD
#The load element is used to model by default constant active and reactive power consumption. We create a 2 MW / 4 MVar load with a scaling factor of 0.6:
load1 = pp.create_load(net, bus1, p_mw=50, q_mvar=30.99, name="load 1")
load2 = pp.create_load(net, bus2, p_mw=170, q_mvar=105.35, name="load 2")
load3 = pp.create_load(net, bus3, p_mw=200, q_mvar=123.94, name="load 3")
load4 = pp.create_load(net, bus4, p_mw=80, q_mvar=49.58, name="load 4")
net.load

#VOLTAGE CONTROLLED GENERATOR
#The generator element is used to model voltage controlled active power generation. We define it with a active power generation (negative) and a voltage set point:
pp.create_sgen(net, bus4, p_mw=318, vm_pu=1.02, name="generator") 
net.gen

#RUN A POWER FLOW
#Runing a loadflow adds seperate result table with the prefix 'res_':
pp.toolbox.create_continuous_elements_index(net)
pp.runpp_3ph(net)
net
#These results tables are pandas datafarmes with the same index as the element table. For example, the bus table contains all bus voltages and summed bus power injections:
net.res_bus_3ph

closed 2026-11-09 by vogt31337