e2nIEE/pandapower #1229: Voltage Rating for 33/11kV and 11/0.4 kV requirements

1998nina, 2021-05-19:
Hi! I'm still new to python programming and I was wondering if someone have the requirements for voltage transformer rating of 33/11kV and 11/0.4kV which are 
1) vkr_percent - real part of relative short-circuit voltage 
2) vk_percent - relative short-circuit voltage 
3) pfe_kw - iron losses in kW 
4) i0_percent - open loop losses in percent of rated current 
5) vk0_percent - zero sequence relative short-circuit voltage 
6) vkr0_percent - real part of zero sequence relative short-circuit voltage 
7) mag0_percent - zero sequence magnetizing impedance/ vk0 
8) mag0_rx - zero sequence magnitizing R/X ratio 
9) si0_hv_partial - Distribution of zero sequence leakage impedances for HV side 
9) shift_degree - Angle shift over the transformer* 
10) tap_side - position of tap changer (“hv”, “lv”)
11) tap_max  - maximal allowed tap position
12) tap_min - minimal allowed tap position
13) tap_neutral - tap position where the transformer ratio is equal to the ratio of the rated voltages
14) tap_step_degree - tap step size for voltage angle in degree*
15) tap_step_percent - tap step size for voltage magnitude in percent
16) tap_phase_shifter - whether the transformer is an ideal phase shifter*

I did a lot of research about the data sheet transformer rating 33/11kV but it turns out the data sheet does not have this requirement. It is very devastating for me because it will delay my progress for my project. It would be helpful if someone would replied this. Thank you!

rbolgaryn, 2021-05-21:
Hi @1998nina ,

Is it correct, that you found the parameters for the transformers in the data sheet, but not exactly the variables you need in pandapower?

rbolgaryn, 2021-05-28:
I think you should be able to calculate the values of the parameters you need from the parameters you have.

closed 2021-06-10 by rbolgaryn