e2nIEE/pandapower #1235: Question: Automatic switching on or off of generators

Crueda96, 2021-05-21:
Sorry if the question is a bit silly, but I'm just learning to occupy the module..
There is some way to automatically turn generators on / off according to the convergence of the flow, that is, if there is a lot of generation for the current load, to be able to disconnect certain generation units; or if there is a generation deficit to supply the load to be able to incorporate generation units?

rbolgaryn, 2021-05-21:
Hi @Crueda96 ,

you can use control module for a task like that.

You can define a custom controller that chacks the total generation for the time steps and adjusts the in_service column of the generators, or the scaling factors. 

Roman

Crueda96, 2021-05-21:
Thanks for the reply @rbolgaryn. Custom controller can be used in systems that do not use timeseries? Because when talking about automatically turning on / off a generator I wanted something like a dispatch of units for a specific hour

rbolgaryn, 2021-05-21:
Controllers can be used also without time series simulation, if you use runpp with run_control=True.

But if you mention a specific hour, then it is part of time series simulation?

Crueda96, 2021-05-21:
OK, thanks a lot. Of course, that specific time is obtained from an analysis of a time series

rbolgaryn, 2021-05-24:
You can read more about creating a controller in this tutorial:

https://github.com/e2nIEE/pandapower/blob/develop/tutorials/building_a_controller.ipynb

Crueda96, 2021-07-05:
@rbolgaryn  I am trying to replicate the code given in the tutorial, the script is not running properly and I receive the following error:
`TypeError:` time_step() takes 2 positional arguments but 3 were given`