e2nIEE/pandapower #1236: BUG Short Circuit Computation with calc_sc()

giovannigasparini, 2021-05-25:
Hello, we probably encountered a bug inside the short circuit calculation performed by `calc_sc()`.
We attach a pandapower network that exhibits the problem, together with a short notebook to execute it.
[CALC SC INDEX ERROR BUG.zip](https://github.com/e2nIEE/pandapower/files/6538258/CALC.SC.INDEX.ERROR.BUG.zip)

Briefly, I try to explain the problem:

- **pandapower\shortcircuit\results.py** `in _get_line_all_results(net, ppc, bus)` at line 106: 
`-net.res_line_sc["ikss_ka"] = minmax(ppc["internal"]["branch_ikss_f"][f:t, ppc_index].real.reshape(-1, 1)` raises an  Index error: _IndexError: index 3348 is out of bounds for axis 1 with size 3348_

- We found that there is no matching between the dimensions of `ppc["internal"]["branch_ikss_f"]` and the indexing given by `ppc_index`. This mismatching is the cause of the error, though we did not find the reason why it happens.

The same pandapower network does not incur into any error when used to compute the powerflow calculation with `run_timeseries()`.

We don’t know if it can be of any help, but we noticed the following behaviour:

- We found that the computation for the network in question does flow into the following line of code inside `_get_bus_ppc_idx_for_br_all_results`:  
`ppc_index[ppc["bus"][bus_lookup[ppc_index], BUS_TYPE] == 4] = -1` 
There, multiple elements are set to -1. 
Using other networks, that do not raise errors, we found that the case of multiple elements set to -1 does not happen. 

Can you give us some feedback?

wangzhenassd, 2021-05-25:
Hi @giovannigasparini,
thank you very much for reporting this and offering the example, which can reproduce the issue! I will take a look on that.

wangzhenassd, 2021-05-27:
Hi @giovannigasparini, can you try the example with the latest develop branch? In your code rdss_pu needs to be converted to rdss_ohm.

giovannigasparini, 2021-05-27:
Hi @wangzhenassd . I try to run the example on the dev branch. 
What do you mean with 

> "In your code rdss_pu needs to be converted to rdss_ohm"

?
Shall I just rename the columns *rdss_pu*  into *rdss_ohm* in the Gen DataFrame or should I apply also a formula to convert the data? Thank you for the feedback

wangzhenassd, 2021-05-27:
Yes, just use a formula to convert pu -> ohm. Furthermore, I think if you are considering sgen in LV/MV network maybe it's better to model it as sgen (current source).

giovannigasparini, 2021-05-27:
Ok. I tried the example on the develop branch and there is no error. 
Is there any schedule about releasing an update on the master branch?
Thank you @wangzhenassd

wangzhenassd, 2021-05-27:
Yes, actually, a new release is planned in June

giovannigasparini, 2021-09-28:
I am closing the issue, since the fix has been released. Thank you for the support!

closed 2021-09-28 by giovannigasparini