e2nIEE/pandapower #1239: short circuit (3ph) for a/sync gen/motors

daniel-env, 2021-05-26:
I am reading the vde0102 and I compare it with your documentation/implementation and have the following issues. I hope this is the right place to discuss this.

Issue 1:  
For synchronous generators your documentation (https://pandapower.readthedocs.io/en/v2.6.0/shortcircuit/voltage_source.html) as well as your code says 𝑉𝑁,𝑔𝑒𝑛 divided by 𝑉𝑁,𝑏𝑢𝑠 is the right relation to use for the correction factor. But in VDE0102 the relation is reversed. I am assuming that the VDE0102 is right, because if the rated generator voltage is higher the contribution to the short circuit current must be higher, too. Thus, the short circuit impedance and this correction factor must be smaller. If I am right, maybe this only rarely resulted in errors because 𝑉𝑁,𝑔𝑒𝑛 is mostly very close to 𝑉𝑁,𝑏𝑢𝑠. 

Issue 2:  
Your documentation for asynchronous motors states that "k is the ratio of nominal to short circuit current". But I would argue that it should say "ratio of short circuit to nominal current" as VDE and also your code are using it like I propose (see create_motor()-attribute lrc_pu - locked rotor current in relation to the rated motor current).

Issue 3:
This one might be a bit more complex. I am looking at the shortcircuit-impedance of asynchronous motors in `pandapower/pandapower/build_bus.py: _add_motor_impedances_ppc()`. I understand and agree how you calculate the impedance in [Ohm]. But after this, you are dividing it by the base impedance, making the value per Unit and adding the per unit admittance values to the ppc-matrix. More precisely, you are adding the values as `shunt conductance gs` and `shunt inductance bs`. My problem is that I think that gs and bs have the unit MVA because its the power demanded at V = 1.0 p.u.. There are two reasons. First, to calculate the Y-Bus for the power flow you divide gs and bs with the base-power (Sbase) (pandapower/pandapower/pf/makeYbus_numba.py: line150). And I think the values in the ppc-Y-bus should be per unit. Secondly,  if you compare this with the calculation for synchronous generators in the same file, you also use per unit values mostly, but you divide by sn_gen at the end to get z_gen. So I argue that gs and bs should have the same unit for asynchronous motors as for the synchronous generators. If I am correct and you are missing to divide Z_gen by S_gen, than your Z_gen is way too high and the short circuit current way to low. I created a test case and created one case with a sync generator and another case with an async motor instead. I tried to give them about the same attributes and the results are as I assumed. But maybe I am wrong and then I would really appreciate your thoughts were I went wrong :)

HakonGabrielsen, 2021-06-09:
I also discovered Issue 3. The problem to me seems to be that calculation of z_motor_pu in line 583 of _add_motor_impedances_ppc() in build_bus.py does not include the baseMVA of the system, which makes the per unit value referred to 1 MVA as base, not the actual system baseMVA value. In my network, using 1000 MVA resulted in too low short circuit contribution from motors, while changing sn_mva to 1 MVA gave correct values. Either the z_motor_pu calculation should divide by ppc['baseMVA'], or gs and bs should be multiplied by ppc['baseMVA'] before being put to ppc["bus"][buses, GS] in lines 589-591. The latter would be just like lines 557-559 in _add_ext_grid_sc_impedance(). 

I think another issue is that gs and bs are assigned directly to ppc, both in lines 557-558 and in lines 590-591. I think this overwrites any admittance of other elements on the same bus. It seems like having a motor on the same bus as an ext_grid element results in the motor admittance overwriting the grid admittance, effectively removing the short circuit contribution from the external grid. If my understanding is correct, this could be resolved by replacing "=" by "+=" when assigning to ppc.

lthurner, 2021-06-11:
Thanks guys for double checking the calculation in detail, much appreciated!

There are multiple issues adressed here, I am still trying to get an overview:

1. Vgen/Vbus vs. Vbus/Vgen: I would have to check the literature again, but when I reverse the voltage fraction, the tests are failing. The tests match results from PowerFactory, which I would assume to be correct.
2. Documentation: you are right, this is just a wrong formulation in the docs
3. Yes it seems like @HakonGabrielsen is right, the issue here is that the baseMVA calculation is missing. The tests are all carried out with baseMVA of 1, which is why it has not been discovered before. 
4. It is correct that the admittance should be added instead of overwritten. The same goes for gen and ext_grid.

I will put in a PR fixing these issues. @wangzhenassd do you have an idea about Issue 1?

daniel-env, 2021-06-15:
Hi I can confirm that issue3 is solved for me.

Regarding Issue1: yes you  are right that  kg = vn_net/(vn_gen ...) is already corrected in the develop. But in the docs its still the other way round?

daniel-env, 2021-06-15:
And out of curiosity:
can you estimate how long will it take to make a new release with the current develop?

wangzhenassd, 2021-06-16:
Hi @daniel-env, thank you for the feedback. The doc will be updated shortly, a release is planned in the next two weeks.

SteffenMeinecke, 2021-11-11:
As far as I can see, all issues are covered and the release happend. If I am wrong, @daniel-env @HakonGabrielsen  please reopen the issue and state which part wasn't fixed.

closed 2021-11-11 by SteffenMeinecke