e2nIEE/pandapower #1246: running opf in timeseries

angelinasyrri, 2021-06-01:
Dear pandapower team, 

I am currently trying to use the timeseries module with opf with the following statement:

pp.timeseries.run_timeseries(net, time_steps = range(0,24),runopf=pp.runopp(net,delta=1e-16))

It seems to work, however, I do not see in the documentation anywhere that timeseries supports opf apart from pf. Can anybody confirm that what I doing makes sense?

Best

rbolgaryn, 2021-06-02:
Dear @angelinasyrri ,

I don't see how your code would work.
The functionality can be used with the parameter run=pp.runopp

We are about to add a tutorial to cover that topic, I will link it here when it is ready.

Roman

angelinasyrri, 2021-06-02:
Hello Roman, 

Thanks for the prompt reply. I am looking forward to the relevant tutorial!

Just to make sure I understood correctly, do you mean that you cannot confirm that the code would work since I did not provide any output, but timeseries functionality can indeed be used with opf with the aforementioned parameter? Or that this combination wouldn't work, no matter what?

Best

rbolgaryn, 2021-06-02:
Hi @angelinasyrri ,

here the Pull Request for the tutorial with an example:

https://github.com/e2nIEE/pandapower/pull/1251

I think the code you provided would not work. Using the "run" parameter, as in the tutorial in the pull request, should work.

I am going to merge it in a moment and then you can just update your pandapower from the develop branch and take a look at the example. Please let me know if it works for your use-case!

Best regards,
Roman

angelinasyrri, 2021-06-03:
Hey! Many thanks for the tutorial and the fast reply. I 'll move on with the proper parameter and let you know once I confirm proper results! Cheers:)

closed 2021-06-11 by lthurner