e2nIEE/pandapower #1249: Optimal Power Flow

K3ding, 2021-06-02:
Dear Pandapower Team,

i'm currently writing my bachelor thesis and I'm trying to solve an OPF calculation for a network with a changing load (from csv file), changing renewable sources (also from csv files), and fixed power fossil power plants. When I'm building a simple island network the OPF converts for every timestamp. But when I add an external grid, my OPF does not convert.

This is my Network Program:

'Create empty Network'
    net = pp.create_empty_network()
    'Create Buses'
    b0 = pp.create_bus(net, max_vm_pu=1.1, min_vm_pu=0.9, name='0', type='b', vn_kv=10, zone='1') #connection bus: ext_grid -> gen/load
    b1 = pp.create_bus(net, max_vm_pu=1.1, min_vm_pu=0.9, name='2', type='b', vn_kv=10, zone='1') #generator 1 bus
    b2 = pp.create_bus(net, max_vm_pu=1.1, min_vm_pu=0.9, name='3', type='b', vn_kv=10, zone='1') #generator 2 bus
    b3 = pp.create_bus(net, max_vm_pu=1.1, min_vm_pu=0.9, name='4', type='b', vn_kv=10, zone='1') #generator 3 bus
    b4 = pp.create_bus(net, max_vm_pu=1.1, min_vm_pu=0.9, name='5', type='b', vn_kv=10, zone='1') #generator 4 bus
    b5 = pp.create_bus(net, max_vm_pu=1.1, min_vm_pu=0.9, name='5', type='b', vn_kv=10, zone='1') #load bus
    b6 = pp.create_bus(net, max_vm_pu=1.1, min_vm_pu=0.9, name='6', type='b', vn_kv=10, zone='1')   #ext_grid bus
    'Create Lines'
    l1 = pp.create_line(net, from_bus=b0, to_bus=b1, length_km=10, std_type='679-AL1/86-ST1A 110.0',   max_loading_percent=110) #g1 to connection
    l2 = pp.create_line(net, from_bus=b0, to_bus=b2, length_km=10, std_type='679-AL1/86-ST1A 110.0', max_loading_percent=110) #g2 to connection
    l3 = pp.create_line(net, from_bus=b0, to_bus=b3, length_km=10, std_type='679-AL1/86-ST1A 110.0', max_loading_percent=110) #g3 to connection
    l4 = pp.create_line(net, from_bus=b0, to_bus=b4, length_km=10, std_type='679-AL1/86-ST1A 110.0', max_loading_percent=110) #g4 to connection
    l5 = pp.create_line(net, from_bus=b0, to_bus=b5, length_km=10, std_type='679-AL1/86-ST1A 110.0', max_loading_percent=110) #load to connection
    l6 = pp.create_line(net, from_bus=b0, to_bus=b6, length_km=10, std_type='679-AL1/86-ST1A 110.0', max_loading_percent=110) #ext_grid to connection
    'Create External Grid'
    e1 = pp.create_ext_grid(net, bus=b6, vm_pu=1, min_p_mw=-10, max_p_mw=10)
    'Create Generators'
    g1 = pp.create_gen(net, bus=b1, p_mw=0, min_p_mw=0, max_p_mw=5, name='Coal', controllable=True)   #Fuel: Coal
    g2 = pp.create_gen(net, bus=b2, p_mw=0, min_p_mw=0, max_p_mw=2, name='Gas', controllable=True)    #Fuel: Gas
    g3 = pp.create_gen(net, bus=b3, p_mw=0, min_p_mw=0, max_p_mw=5, name='Solar', controllable=True)    #Fuel: Solar
    g4 = pp.create_gen(net, bus=b4, p_mw=0, min_p_mw=0, max_p_mw=5, name='Wind', controllable=True)     #Fuel: Wind
    'Create Loads'
    pp.create_load(net, bus=b5, p_mw=0.1, min_p_mw=0.1, max_p_mw=0.1, controllable=False)

    return net

This is my for-Loop for the OPF:

    for i in range (0, len(df_load)):                            #TODO len(df_load)
        net.gen.iat[2, 13] = df_renewable.iat[i, 0]     #overwrite p_max of solar
        net.gen.iat[3, 13] = df_renewable.iat[i, 1]     #overwrite p_max of wind
        net.load.iat[0, 2] = df_load.iat[i, 0]          #overwrite p_mw of load

        pp.runopp(net)          #verbose=True
        'Results Generators'
        results_smear = net.res_gen['p_mw']                 #p_mw
        results_gen.append(results_smear)
        print('RES_GEN')
        print(results_smear)
        'Results Loads'
        results_smear = net.res_load['p_mw']                #p_mw
        results_load.append(results_smear)
        print('RES_LOAD')
        print(results_smear)
        'Results External Grid'
        results_smear = net.res_ext_grid['p_mw']            #p_mw
        results_ext_grid.append(results_smear)
        print('RES_EXT_GRID')
        print(results_smear)
        'Results Lines'
        results_smear = net.res_line['loading_percent']    #loading_percent
        results_line.append(results_smear)

I hope this is understandable and maybe someone can give me tips on how to solve this :)
Thanks!

rbolgaryn, 2021-06-03:
Hi @K3ding ,

what do you mean by this - "But when I add an external grid, my OPF does not convert."

Also, please keep in mind that we do not have time to answer a question about why an OPF calculation does not converge. You would need to debug the grid and make sure all elements are connected properly and all parameters are correct. Also, it is very important to check if the constraints would allow a solution to be found. For example, you can manually adjust all the loads/generators and see if it solves the issues in the grid. It can often happen, that the constraints are not compatible with a feasible OPF solution.

Roman

K3ding, 2021-06-07:
Hi @rbolgaryn 

Thanks for the reply. I gues a better asked question would be:

How could a OPF converge but when I change the €/MW it does not converge. So basically all the data stays the same (p_mw, constraints etc.) but I only change the price and then my OPF does not converge. Is there a clear cut way to fix this problem?

Thanks a lot!

rbolgaryn, 2021-06-07:
Hi @K3ding ,

this issue could be caused by some numerical problems, I unfortunately cannot give a more precise answer.
You can try modifying the OPF parameters for the tolerance of calculations and see if you achieve convergence even with the changed costs. You can also try setting all costs in ct/MWhr and see if it makes a difference. You can also modify the base MVA in net.sn_mva and set it to a higher value, e.g. 10, 100, 1000. 

from the documentation, the optional parameters for OPF are:

```python
  kwargs - Pypower / Matpower keyword arguments:

      OPF_VIOLATION (5e-6) constraint violation tolerance

      PDIPM_COSTTOL (1e-6) optimality tolerance

      PDIPM_GRADTOL (1e-6) gradient tolerance

      PDIPM_COMPTOL (1e-6) complementarity condition (inequality) tolerance

      PDIPM_FEASTOL (set to OPF_VIOLATION if not specified) feasibiliy (equality) tolerance

      PDIPM_MAX_IT (150) maximum number of iterations

      SCPDIPM_RED_IT(20) maximum number of step size reductions per iteration
```

Best regards, 
Roman

rbolgaryn, 2021-06-24:
Dear @K3ding ,

I am closing the issue, if you have further questions, feel free to reopen it or start a new thread in Discussions.

https://github.com/e2nIEE/pandapower/discussions

Roman

closed 2021-06-24 by rbolgaryn

Adelheide, 2022-02-10:
Hello @K3ding , I was wondering if you could help me. Would it be possible for you to tell me how you have defined the changing load in the script? because I understand that it is a load of which you have different measurements during time, right?

In my case I have a bus where the load that comes from there is changing with time, and I have different data for this load such as the current, power, tension, etc (versus time) in .json and .csv. Did you have something like that? 

Furthermore, do you know if the .csv must have a determined internal configuration of the parameters that you have of the changing load?

Thank you ;)