e2nIEE/pandapower #1268: One problem with IEEE 33bus system

hitchenzhe, 2021-06-17:
Hi,

I found net = pn.case33bw() is a meshed network? But normally, it should be radial.

Any solution to that?

Best,

Zhe

wangzhenassd, 2021-06-18:
What exactly do you mean by meshed?

SteffenMeinecke, 2021-06-21:
@hitchenzhe The network `pn.case33bw()` was converted from the matpower case. It includes no mesh/loop.

wangzhenassd, 2021-06-24:
@hitchenzhe Is this question cleared?

hitchenzhe, 2021-06-24:
> mesh

Actually, if you plot the net topo, you could find there exist loops, the reason is switch line data included in net.line (index 32-36). Yes, net dataframe has no switches, I am not sure whether the power flow result consider the switch line.

wangzhenassd, 2021-06-24:
@hitchenzhe There is lines out-of-service in the network, by ignoring these the network is not meshed.

hitchenzhe, 2021-06-25:
> @hitchenzhe There is lines out-of-service in the network, by ignoring these the network is not meshed.

OK, fair enough, but why not separate net.line into switch type and line type. May be more clear for usage:) When I first see the topo, it is amazing that there exists loop.

Thanks a lot.
Zhe

hitchenzhe, 2021-06-25:
> @hitchenzhe There is lines out-of-service in the network, by ignoring these the network is not meshed.

By the way, do you know how to get Ybus matrix of specific network? Like IEEE 33.

wangzhenassd, 2021-06-27:
Hi @hitchenzhe, I think it is because these branches already exist but simply set as out of service in the original dataset.

 Ybus can be found in "net["_ppc"]["internal"]["Ybus"]" after AC powerflow is finished.

BinHuangScut, 2021-07-01:
Hi Zhenqi,
Thank you for your efforts on pandapower. 
I would like to ask if you have any plans to release the code related to droop control in the near future, or if you can share the code of droop control implemented by control module. 

Thank you, 
Bin

hitchenzhe, 2021-07-07:
> Hi @hitchenzhe, I think it is because these branches already exist but simply set as out of service in the original dataset.
> 
> Ybus can be found in "net["_ppc"]["internal"]["Ybus"]" after AC powerflow is finished.

Yeah, but I found another issue in this matrix. take IEEE 33 bus as an example, there is a value (-511.5021118640935+488.11024699225993j) at row 10, col 1. While there is no link between bus 2 and bus 11!

SteffenMeinecke, 2021-11-11:
> When I first see the topo, it is amazing that there exists loop.

@hitchenzhe what topo did you see? If I use the following, I see no mesh:
```
import pandapower.networks as pn
import pandapower.plotting as plot
net = pn.case33bw()
plot.simple_plot(net)
```

SteffenMeinecke, 2021-11-11:
> Yeah, but I found another issue in this matrix. take IEEE 33 bus as an example, in Ybus matrix, there is a value (-511.5021118640935+488.11024699225993j) at row 10, col 1. While there is no link between bus 2 and bus 11!
@hitchenzhe Is the following what you meant?:
```
import pandapower as pp
import pandapower.networks as pn
net = pn.case33bw()
pp.runpp(net)
Ybus = net["_ppc"]["internal"]["Ybus"]
print(Ybus[10, 1])
```
The output is `0j`.

SteffenMeinecke, 2021-11-11:
If a problem remains, please reopen.

closed 2021-11-11 by SteffenMeinecke