e2nIEE/pandapower #1269: Large error in state estimation

tyua321, 2021-06-19:
I tried to use state estimation on a 14-bus system, however the error is very large here. the MSE error for the voltage angle is more than 8, usually the error should be something like 0.0001. In addition when I tried this on 118-bus system, the estimation failed.  Could you please help me find is there something wrong with the estimation part? My code is followed. Thank you!
```
error1=0.01
error2=0.02
net14=pn.case14()
pp.runpp(net14)
bus_data=net14.res_bus.sort_index().to_numpy()
for i in range(bus_data.shape[0]):
        v=bus_data[i,0]+np.random.normal(0, np.absolute(1*error1))
        p=bus_data[i,2]+np.random.normal(0, np.absolute(1*error2)) 
        q=bus_data[i,3]+np.random.normal(0, np.absolute(1*error2))

        pp.create_measurement(net14, "v", "bus", v, error1*1, element=i)
        pp.create_measurement(net14, "p", "bus", p, error2*1, element=i)
        pp.create_measurement(net14, "q", "bus", q, error2*1, element=i)
success = estimate(net14, init="flat")
net14.res_bus_est=net14.res_bus_est.sort_index()
V, theta = net14.res_bus_est.vm_pu, net14.res_bus_est.va_degree
p1=bus_data[:,0]-V.to_numpy()
p2=bus_data[:,1]-theta.to_numpy()
loss_v+=sum(p1**2)/14
loss_theta=sum(p2**2)/14
```

wangzhenassd, 2021-06-19:
Hi, by simply adjusting your std_dev to error1=0.001, error2=0.2, I got the result in plausible range. As for case118, which is a much larger network, you need to add measurements for lines/trafos. Generally you need to have 4 x number of buses measurements to make WLS work properly.

[image]

tyua321, 2021-06-19:
But in my case, even the std is changed to 0.001, the result remained same. It seems like the result is not improving when the std is decreased. 
```
import pandapower.networks as pn
import pandapower as pp
import numpy as np
from pandapower.estimation import estimate

error1=0.001
error2=0.001
net14=pn.case14()
pp.runpp(net14)
bus_data=net14.res_bus.sort_index().to_numpy()

for i in range(bus_data.shape[0]):
        v=bus_data[i,0]+np.random.normal(0, np.absolute(error1))
        p=bus_data[i,2]+np.random.normal(0, np.absolute(error2)) 
        q=bus_data[i,3]+np.random.normal(0, np.absolute(error2))

        pp.create_measurement(net14, "v", "bus", v, error1, element=i)
        pp.create_measurement(net14, "p", "bus", p, error2, element=i)
        pp.create_measurement(net14, "q", "bus", q, error2, element=i)
success = estimate(net14, init="flat")
net14.res_bus_est=net14.res_bus_est.sort_index()
V, theta = net14.res_bus_est.vm_pu, net14.res_bus_est.va_degree
p1=bus_data[:,0]-V.to_numpy()
p2=bus_data[:,1]-theta.to_numpy()
loss_v=sum(p1**2)/14
loss_theta=sum(p2**2)/14
```
[image]

wangzhenassd, 2021-06-20:
The point is your v measurements should be much smaller std_dev as pq measure, if you want accurate v value. such as 0.001 for v and 0.2 for pq used in my example. or else you should check the accuracy of bus pq

tyua321, 2021-06-20:
But I still found it strange here. I saw some paper that also used 0.01 as deviation for all the measurement in 14-bus network, the error for angle is about 0.01. In my case the error is too large, almost 20%. Do you have any idea why this happened?

wangzhenassd, 2021-06-20:
can you share the links to the papers?  The behaviour could also relate to the initialization.

tyua321, 2021-06-20:
https://sie.scholasticahq.com/article/4640-power-system-state-estimation-using-microsoft-excel
https://onlinelibrary.wiley.com/doi/pdf/10.1002/tee.20692

tyua321, 2021-06-20:
Also, I found the Matlab code for state estimation here
https://www.mathworks.com/matlabcentral/fileexchange/23052-power-system-state-estimation-using-wls
It  has 14-bus example here,  I set all the deviations to 0.01, both voltage and angle error are much smaller compared to my pandapower results.

wangzhenassd, 2021-06-20:
Thank you for offering the examples! I tried also the matlab code with simply changing the test network to ieee case14, it gives the following vm,va results:
[image]
While pandapower gives me the following results, by setting std_dev to 0.01 for both v and pq:
[image]
The only adaption I made to your code is to make the std_dev relativ to the measurement, since that is the resulted error when std_dev is supposed to be 1%. I see that pandapower estimation and matlab estimation gives actually similar results with some deviation can be observed on vm.
```python
for i in net14.bus.index:
        v=bus_data.at[i,"vm_pu"]+np.random.normal(0, np.absolute(1*error1))
        p=bus_data.at[i,"p_mw"]+np.random.normal(0, np.absolute(1*error2)) * bus_data.at[i,"p_mw"]
        q=bus_data.at[i,"q_mvar"]+np.random.normal(0, np.absolute(1*error2)) * bus_data.at[i,"q_mvar"]

        pp.create_measurement(net14, "v", "bus", v, error1*1, element=i)
        pp.create_measurement(net14, "p", "bus", p, max(0.01, error2*abs(p)), element=i)
        pp.create_measurement(net14, "q", "bus", q, max(0.01, error2*abs(q)), element=i)
```

And the real powerflow results at the end:
[image]

wangzhenassd, 2021-06-24:
Hi @tyua321, is your question cleared?

tyua321, 2021-06-24:
Yes, thank you for your time, I will close the question now.

closed 2021-06-24 by tyua321