e2nIEE/pandapower #1273: Inconsistent result tables of asymm. elements in 3ph power flow

hakanozlemis, 2021-06-22:
## Problem:
Result tables for "asymmetric loads" and "asymmetric sgens" (`res_asymmetric_load_3ph` and `res_asymmetric_sgen_3ph`) are being created wrong in 3-phase power flow simulations if the element indices in the network instance are not sorted and don't have sequential indices (in range of [0, (#elements - 1)]). 
* `NaN` values appear as active and reactive power for elements with indices higher than `(#elements - 1)` although they have initially relevant values. 
* Furthermore, the power values of different elements are being shuffled such that a load gets power value of an another load with a different index.  

Note: This problem does not cause any bug with power flow calculations and the other result tables since the problem is only at results extraction of controllable asymmetric elements ("asymmetric_load" and "asymmetric_sgen")

## How to reproduce
- [ ] Import MVOberrhein network
- [ ] Make all network elements ready for 3-phase power flow
- [ ] Add an additional asymmetric load or change all loads to asymmetric_load
- [ ] Change index of one asymmetric load to 10000 (which is higher than (#asymmetric_loads- 1))
- [ ] Run 3-phase power flow
- [ ] See `res_asymmetric_load_3ph` for the asymmetric load with changed index

## Reason:
The function `write_pq_results_to_element_3ph` in `results_bus.py` is the reason of this bug. It writes the extracted data `el_data` into result tables by creating a new `pandas.Series` from list of values. Creating a new `pandas.Series` from list of values creates itself new indices for `pandas.Series`, starting from `0` to `(#elements - 1)`. Writing a `pandas.Series` into a `pandas.DataFrame` with different indices causes this bug. The issue can be represented by:
```
import pandas as pd

data = {'col1': [1, 2], 'col2': [3, 4]}
df = pd.DataFrame(data=data, index=[10, 0])
print(df, end='\n\n')

ds = pd.Series([5, 15])
print(ds, end='\n\n')

df['col1'] = ds
print(df)
```

## Possible solution:
- [ ] Use lists instead of using `pandas.Series` for updating result tables

rbolgaryn, 2021-06-24:
Hi @hakanozlemis ,

thank you very much for the detailed and comprehensive bug report!

We will address this issue promptly.

Roman

rbolgaryn, 2021-07-14:
Hi @hakanozlemis ,

we are on the issue, however it will take some time to fix it.

Roman

elodin, 2021-10-04:
Thanks for the detailed issue report @hakanozlemis!

I could recreate the NaN problem. 
The shuffling of values was not happening on my tests, however. Do you have the possibility to put the code leading to the issue online so i can reproduce it? 

The NaN issue has been solved by changing the datatype from pandas.series to list, as proposed.

hakanozlemis, 2021-10-06:
What I meant with the shuffling of values, was exemplarily that the order of the value 5 in the simple example above is changed from first row to second row. It was expected that the elements with IDs 10 and 0 get the values 5 and 15 respectively (5 stays in the first row and the 15 in the secon row). However, the element in the second row (with ID 0) becomes the value 5 since the created series `ds` has unintentionally the index 0 for its first value, which is 5 in our case. I hope this clarifies your question. However, that also should already be solved with the proposed solution.

elodin, 2021-11-19:
@SteffenMeinecke @rbolgaryn @SimonRubenDrauz 
This issue has been addressed together with Gourab. The solution needs to be checked. Please tell me if it should be revised in any way or else it can be merged.

SteffenMeinecke, 2021-11-23:
@elodin @hakanozlemis To remove pandas Series here is ok to avoid the indexing problem. But from my point of view, we can but we don't need to use lists. numpy arrays are already fine. But I **don't** think, that this is big point which should be still be changed.

@hakanozlemis I don't see the second point, too. The results get the indices of the elements, even though the elements indices are weird. For example test with the following test script (which uses `runpp()` instead of `runpp_3ph()` but both use `init_element()` so it should be the same issue):

```
import pandapower as pp
import pandapower.networks as nw

net = nw.create_cigre_network_mv()
pp.runpp(net)
print(net.res_bus)
pp.reindex_buses(net, dict(zip(net.bus.index.values,
                               [4, 2, 1, 3, 6, 7, 8, 9, 15, 12, 18, 17, 16, 25, 21])))
pp.runpp(net)
print("results are the same and the indices fit to the reindexed buses:")
print(net.res_bus)
```

If @hakanozlemis doesn't clarify the open issue, from my point of view, the PR can be confirmed and this issue be closed.

hakanozlemis, 2021-11-23:
@SteffenMeinecke As mentioned earlier, the 3-phase result tables of ``asymmetric_load``'s and ``asymmetric_sgen``'s are not correct (``res_asymmetric_load_3ph`` and ``res_asymmetric_sgen_3ph`` correspondingly) which are filled by `runpp_3ph()`. The indexing problem does not occur with `runpp()`.

@elodin Sorry for my late response. I could test your fix now and it works well and solves the both issues I mentioned. Here is my test case:

``` python 
import pandapower as pp

print(pp.__version__)

net = pp.create_empty_network('2bus_net')

b0 = pp.create_bus(net, vn_kv=20.0, name='bus0', index=0)
b1 = pp.create_bus(net, vn_kv=20.0, name='bus1', index=1)

pp.create_ext_grid(net, 1, voltage=1.03, s_sc_max_mva=1000, rx_max=0.1, r0x0_max=0.1, x0x_max=1.0)

pp.create_line(net, b0, b1, length_km=1, std_type='243-AL1/39-ST1A 20.0')
net.line['r0_ohm_per_km'] = 0.01
net.line['x0_ohm_per_km'] = 0.01
net.line['c0_nf_per_km'] = 0.01

pp.create_asymmetric_sgen(net, b0, p_a_mw=4, index=3)
pp.create_asymmetric_sgen(net, b0, p_a_mw=5, index=0)
pp.create_asymmetric_sgen(net, b0, p_a_mw=6, index=1)

pp.create_asymmetric_load(net, b1, p_a_mw=15)

pp.runpp_3ph(net)
print(net.res_asymmetric_sgen_3ph)
```  

I want to also add the outputs of the runs with versions 2.7.0 (latest release) and your branch.  
Firstly, the output of latest release:  
``` python
2.7.0
   p_a_mw  q_a_mvar  p_b_mw  q_b_mvar  p_c_mw  q_c_mvar
3     NaN       NaN     NaN       NaN     NaN       NaN
0     4.0       0.0     0.0       0.0     0.0       0.0
1     5.0       0.0     0.0       0.0     0.0       0.0
```
As you can see here, for the asymmetric sync. generator, the result table is filled with 'NaN' (the first issue mentioned at the beginning). And secondly, the result table has `p_a_mw=4.0` for the generator 0, however this should have been the power value of generator with index 3. Furthermore, the generator 1 has the power value of generator 0.  This is what is meant with 'shuffling' of power values. This is caused again by the ``pandas.Series``. Writing a ``pandas.Series`` into a ``pandas.DataFrame`` with different indices causes this bug. 

And the output after your fix:
``` python
2.7.1
   p_a_mw  q_a_mvar  p_b_mw  q_b_mvar  p_c_mw  q_c_mvar
3     4.0       0.0     0.0       0.0     0.0       0.0
0     5.0       0.0     0.0       0.0     0.0       0.0
1     6.0       0.0     0.0       0.0     0.0       0.0
````
Here is everything fine. Your fix works well. Thanks!

closed 2021-11-25 by SteffenMeinecke