e2nIEE/pandapower #1290: Convergence problems on large net models

alexlovushkin, 2021-07-08:
Hi all,

I’m testing state estimation capabilities of PandaPower. To emulate measurements i pass power flow results as feedback. For small test grids(case9, case30, case39) all works fine, estimation is precise and bad data detection function robust to distortions in voltage, injection and flow meases. But when I try to apply this method to a little more larger networks (case89pegase for example) both basic WLS and advanced robust estimation algorithms don’t converge to correct point, or even diverges at all:

```"python"
from pandapower.estimation import estimate
from pandapower.networks import *
from math import fabs
import pandapower as pp

def get_net():
    return case89pegase()

def print_net_est_res(net):

    print("\n########################################################################################################")

    print(">>> est_res_bus:")
    print(net.res_bus_est[['vm_pu', 'p_mw', 'q_mvar']])
    print(">>> est_res_trafo:")
    print(net.res_trafo_est[['p_hv_mw', 'q_hv_mvar', 'p_lv_mw', 'q_lv_mvar', 'i_hv_ka', 'i_lv_ka']])
    print(">>> est_res_trafo3w:")
    print(net.res_trafo3w_est[['p_hv_mw', 'q_hv_mvar', 'p_mv_mw', 'q_mv_mvar', 'p_lv_mw', 'q_lv_mvar', 'i_hv_ka', 'i_mv_ka', 'i_lv_ka']])
    print(">>> est_res_line:")
    print(net.res_line_est[['p_from_mw', 'q_from_mvar', 'p_to_mw', 'q_to_mvar', 'i_from_ka', 'i_to_ka']])

def print_est_comparison(net, net2, alarm_thr, noise_lim):
    pd.set_option('display.max_rows', None)
    pd.set_option('display.max_columns', None)
    pd.set_option('display.width', None)

    def diff_stat(ref_val, val, alarm_thr, noise_lim):
        try:
            abs_diff = ref_val - val
        except Exception as ex:
            print(ex)

        if fabs(ref_val) < noise_lim or fabs(val) < noise_lim:
            return '+'
        else:
            rel_diff = fabs(100 * abs_diff / ref_val)
            if rel_diff > alarm_thr:
                return '{:.3e}({:.2f}%)'.format(abs_diff, rel_diff)
            else:
                return '+'

    # bus`s
    for busIndex in net.bus.index:
        net2.res_bus_est.vm_pu[busIndex] = diff_stat(net.res_bus.vm_pu[busIndex], net2.res_bus_est.vm_pu[busIndex], alarm_thr, noise_lim)
        net2.res_bus_est.p_mw[busIndex] = diff_stat(net.res_bus.p_mw[busIndex], net2.res_bus_est.p_mw[busIndex], alarm_thr, noise_lim)
        net2.res_bus_est.q_mvar[busIndex] = diff_stat(net.res_bus.q_mvar[busIndex], net2.res_bus_est.q_mvar[busIndex], alarm_thr, noise_lim)

    # line`s
    for lineIndex in net.line.index:
        net2.res_line_est.p_from_mw[lineIndex] = diff_stat(net.res_line.p_from_mw[lineIndex], net2.res_line_est.p_from_mw[lineIndex], alarm_thr, noise_lim)
        net2.res_line_est.p_to_mw[lineIndex] = diff_stat(net.res_line.p_to_mw[lineIndex], net2.res_line_est.p_to_mw[lineIndex], alarm_thr, noise_lim)

        net2.res_line_est.q_from_mvar[lineIndex] = diff_stat(net.res_line.q_from_mvar[lineIndex], net2.res_line_est.q_from_mvar[lineIndex], alarm_thr, noise_lim)
        net2.res_line_est.q_to_mvar[lineIndex] = diff_stat(net.res_line.q_to_mvar[lineIndex], net2.res_line_est.q_to_mvar[lineIndex], alarm_thr, noise_lim)

        net2.res_line_est.i_from_ka[lineIndex] = diff_stat(net.res_line.i_from_ka[lineIndex], net2.res_line_est.i_from_ka[lineIndex], alarm_thr, noise_lim)
        net2.res_line_est.i_to_ka[lineIndex] = diff_stat(net.res_line.i_to_ka[lineIndex], net2.res_line_est.i_to_ka[lineIndex], alarm_thr, noise_lim)

    # trafo`s
    for trafoIndex in net.trafo.index:
        net2.res_trafo_est.p_hv_mw[trafoIndex] = diff_stat(net.res_trafo.p_hv_mw[trafoIndex], net2.res_trafo_est.p_hv_mw[trafoIndex], alarm_thr, noise_lim)
        net2.res_trafo_est.p_lv_mw[trafoIndex] = diff_stat(net.res_trafo.p_lv_mw[trafoIndex], net2.res_trafo_est.p_lv_mw[trafoIndex], alarm_thr, noise_lim)

        net2.res_trafo_est.q_hv_mvar[trafoIndex] = diff_stat(net.res_trafo.q_hv_mvar[trafoIndex], net2.res_trafo_est.q_hv_mvar[trafoIndex], alarm_thr, noise_lim)
        net2.res_trafo_est.q_lv_mvar[trafoIndex] = diff_stat(net.res_trafo.q_lv_mvar[trafoIndex], net2.res_trafo_est.q_lv_mvar[trafoIndex], alarm_thr, noise_lim)
[... 403 lines cut ...]
82                    +     -5.620e+00(3.67%)                  +     5.533e+00(3.59%)   -2.435e-02(3.78%)   -2.435e-02(3.78%)
83                    +     -6.911e+00(3.06%)                  +     6.848e+00(3.02%)   -2.074e-02(3.35%)   -2.074e-02(3.35%)
84                    +     -9.744e+00(3.47%)                  +                    +   -7.228e-02(4.48%)   -7.228e-02(4.48%)
85                    +     -5.934e+00(2.22%)                  +     5.364e+00(1.96%)   -5.729e-02(4.02%)   -5.729e-02(4.02%)
86                    +     3.657e+00(10.97%)                  +   -3.681e+00(10.96%)   -6.311e-02(4.97%)   -6.311e-02(4.97%)
87                    +      2.699e-02(2.88%)                  +   -6.560e-02(12.68%)   -4.084e-04(4.50%)   -4.084e-04(4.50%)
88                    +      3.123e-01(2.13%)                  +   -9.224e-01(11.27%)   -6.477e-03(4.58%)   -6.477e-03(4.58%)
89                    +      5.335e-01(4.05%)                  +   -2.385e+00(37.93%)   -2.246e-02(4.65%)   -2.246e-02(4.65%)
9      1.249e-02(4.07%)      2.679e-02(7.50%)  -1.286e-02(4.17%)    -2.913e-02(7.94%)  -2.096e-04(11.83%)  -2.096e-04(11.83%)
90                    +      8.060e-01(4.38%)                  +    -8.540e-01(5.09%)   -1.196e-03(1.47%)   -1.196e-03(1.47%)
91                    +      1.215e-01(4.29%)                  +   -2.141e-01(11.82%)   -1.165e-03(4.44%)   -1.165e-03(4.44%)
92                    +      1.250e-01(6.28%)                  +   -2.003e-01(17.42%)   -1.000e-03(4.39%)   -1.000e-03(4.39%)
93                    +    -5.010e+00(13.06%)                  +    4.955e+00(12.66%)   -1.010e-02(3.49%)   -1.010e-02(3.49%)
94                    +     -1.231e+01(5.20%)                  +                    +   -8.278e-02(4.34%)   -8.278e-02(4.34%)
95                    +      7.838e-01(1.20%)                  +    -8.005e-01(1.22%)   -5.527e-02(5.61%)   -5.527e-02(5.61%)
96                    +                     +                  +                    +   -4.789e-02(4.18%)   -4.789e-02(4.18%)
97                    +    -1.384e+00(57.13%)                  +    1.384e+00(57.13%)  -2.256e-03(63.89%)  -2.256e-03(63.89%)
98                    +      5.777e-01(1.89%)                  +    -5.779e-01(1.89%)   -5.775e-03(3.89%)   -5.775e-03(3.89%)
99                    +    -7.609e-01(21.02%)                  +    2.726e-01(23.14%)   -1.370e-02(4.96%)   -1.370e-02(4.96%)
```
It is seen that for a part of the measurement the estimations differs significantly from the PF result. Another observation: for some networks estimations done without using current measuremets is better then with it. I also tried to use PF result as an initial point for SE algo (init=‘results’), however i got the same result. Could enyone explain what i’m doing wrong or give any advice? Is it correct way to use PF result for test SE? Or mayby i use inappropriate meas error weights (stddev) or estimate() function params?

Thanks.

[test.zip](https://github.com/e2nIEE/pandapower/files/6785762/test.zip)

CircleJerkHug, 2021-07-19:
Hi @alexlovushkin ,
I looked up on this as I was dealing with the same issue, I tried chained estimation and warm start for SciPy Optimization Tool. For estimation I added this to code.

```
# Initialize eppci for the algorithm which contains pypower-grid,
# measurements and estimated grid state (initial value)
net, ppc, eppci = pp2eppci(net)

# Initialize algorithm
estimation_wls = WLSAlgorithm(1e-3, 5)
estimation_opt = OptAlgorithm(1e-6, 1000)

# Start Estimation with specified estimator
eppci = estimation_wls.estimate(eppci)
# for some estimators extra parameters must be specified
eppci = estimation_opt.estimate(eppci, estimator="ql", a=3)

# Update the pandapower network with estimated results
net = eppci2pp(net, ppc, eppci)
```

Though, it didn't converged to some cases, but when I changed the estimator argument to "qc" it converged. I tested this till case 6470rte.

marceldipp, 2021-11-15:
Hi @alexlovushkin, your state estimation initialization is correct and the comparison with the power flow calculation results is also valid. I have tested your code with case89pegase and could attribute the incorrect results to the influence of the shunt elements within the grid (same for wards and xwards). To handle the impact of these component types on the SE there are two workarounds. Although the results of the power flow calculation will slightly differ from the results of the Original net, you should be able to get valid results:

You can either set shunt.step = 0 to ignore the shunts in the PF/SE or replace the shunts by loads e.g. with:

```
for bus, p_mw, q_mvar in zip(net.shunt.bus, net.shunt.p_mw, net.shunt.q_mvar):
     pp.create_load(net=net, bus=bus, p_mw=p_mw, q_mvar=q_mvar)
net.shunt.drop((net.shunt.index), inplace=True)
```
Additionally, CircleJerkHug's solution might be sufficient for you.
Thanks for your contribution and for sharing the code.

closed 2021-11-26 by marceldipp