e2nIEE/pandapower #1294: Multiple faults

CircleJerkHug, 2021-07-14:
Hello everyone, 
I am new to pandapower and exploring the functionalities and trying to implement it. Can we implement multiple faults in a circuit? As far I understand, we can induce faults in the net (whole or list of buses) with any one type of fault. At a time, can we have different faults in the new simultenously?
Thank you.

rbolgaryn, 2021-07-15:
Dear @CircleJerkHug ,

In the short-circuit calculation in pandapower, even with the calculation for multiple buses, it is implied that the faults occur only at one bus at a time. In pandapower, calculating several faults that occur simultaneously is not implemented.

What would be your use case for calculating multiple simultaneous faults?

Best regards,
Roman

CircleJerkHug, 2021-07-16:
Hello @rbolgaryn,
Thanks for the feedback, I was trying to induce the faults to simulate cascading fault in the power grid. In a real case scenario, it might be possible that the circuit have multiple simultaneous faults at the same time. I was reading upon this idea in the following research paper and wondered if it can be implemented here.
https://www.sciencedirect.com/science/article/pii/S1110016816000260
Having the idea of the short circuit current under that situation might be a good addition to pandapower. 
Thanks.

rbolgaryn, 2021-07-16:
Hi @CircleJerkHug ,

it sounds interesting.

Feel free to implement it and open a pull request!

Best regards, 
Roman

SteffenMeinecke, 2021-11-11:
Hi @CircleJerkHug are you working on it? Is anyone out there who is interessted in supporting?