e2nIEE/pandapower #1304: Implement distributed slack calculation for grids with separate zones

rbolgaryn, 2021-07-22:
The implementation of distributed slack calculation has been merged:

https://github.com/e2nIEE/pandapower/issues/771

https://github.com/e2nIEE/pandapower/pull/1298

A point that is still open is a calculation in grids that have multiple separate zones.

To this end, the slack variable in newton.py should be an array, and slack weights should be separate arrays per zone. Then the mismatch is adjusted by something like this:

... + s1 * slack_weights1 + s2 * slack_weights2

The Jacobian should receive additional rows, 1 per reference bus in the zone, and a column per zone, where only the slack weights for the zone are included. Then the distributed slack calculation should work also for separate zones.

rbolgaryn, 2021-07-22:
Any help with this implementation will be most welcome.

AnkurArohi, 2022-03-25:
I want to help

vogt31337, 2025-04-02:
will be reponed, when the issue comes up again.

closed 2025-04-02 by vogt31337