e2nIEE/pandapower #1308: Difficulties to set up OPF with PowerModels

rockstaedt, 2021-07-30:
Hey!

I wanted to look into OPF using julia. I set up my environment as instructed in the documentation [here](https://pandapower.readthedocs.io/en/develop/opf/powermodels.html). Everything went fine except step 7 when I run pytest with `python-jl test_powermodels.py` from the corresponding file. I get the following output: 

`signal (11): Segmentation fault: 11
in expression starting at none:0
_ZN4llvm17PMTopLevelManager16addImmutablePassEPNS_13ImmutablePassE at /Users/Eric/opt/miniconda3/envs/general/lib/libLLVM-10.dylib (unknown line)
_ZN4llvm17PMTopLevelManager12schedulePassEPNS_4PassE at /Users/Eric/opt/miniconda3/envs/general/lib/libLLVM-10.dylib (unknown line)
_ZN4llvm17PMTopLevelManager12schedulePassEPNS_4PassE at /Users/Eric/opt/miniconda3/envs/general/lib/libLLVM-10.dylib (unknown line)
_ZN4llvm18PassManagerBuilder25populateModulePassManagerERNS_6legacy15PassManagerBaseE at /Users/Eric/opt/miniconda3/envs/general/lib/libLLVM-10.dylib (unknown line)
ffi_call_unix64 at /Users/Eric/opt/miniconda3/envs/general/lib/libffi.7.dylib (unknown line)
ffi_call_int at /Users/Eric/opt/miniconda3/envs/general/lib/libffi.7.dylib (unknown line)
_ctypes_callproc at /Users/Eric/opt/miniconda3/envs/general/lib/python3.9/lib-dynload/_ctypes.cpython-39-darwin.so (unknown line)
PyCFuncPtr_call at /Users/Eric/opt/miniconda3/envs/general/lib/python3.9/lib-dynload/_ctypes.cpython-39-darwin.so (unknown line)
_PyObject_Call at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyEval_EvalFrameDefault at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyEval_EvalCode at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyFunction_Vectorcall at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyObject_FastCallDictTstate at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
slot_tp_call at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyObject_MakeTpCall at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
call_function at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyEval_EvalFrameDefault at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyFunction_Vectorcall at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
call_function at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyEval_EvalFrameDefault at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyFunction_Vectorcall at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
call_function at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyEval_EvalFrameDefault at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyEval_EvalCode at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyFunction_Vectorcall at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
method_vectorcall at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
call_function at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyEval_EvalFrameDefault at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyFunction_Vectorcall at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
call_function at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyEval_EvalFrameDefault at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyFunction_Vectorcall at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyObject_FastCallDictTstate at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
slot_tp_init at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
type_call at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyObject_MakeTpCall at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
call_function at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyEval_EvalFrameDefault at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyFunction_Vectorcall at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyEval_EvalFrameDefault at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyEval_EvalCode at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyFunction_Vectorcall at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
call_function at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyEval_EvalFrameDefault at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyFunction_Vectorcall at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyObject_FastCallDictTstate at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
slot_tp_init at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
type_call at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyObject_MakeTpCall at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
call_function at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyEval_EvalFrameDefault at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyFunction_Vectorcall at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
call_function at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyEval_EvalFrameDefault at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyEval_EvalCode at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyFunction_Vectorcall at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
object_vacall at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
PyObject_CallFunctionObjArgs at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyObject_GenericGetAttrWithDict at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyEval_EvalFrameDefault at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyFunction_Vectorcall at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
property_descr_get at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyObject_GenericGetAttrWithDict at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyEval_EvalFrameDefault at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
[... 163 lines cut ...]
_PyFunction_Vectorcall at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
call_function at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyEval_EvalFrameDefault at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyFunction_Vectorcall at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
call_function at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyEval_EvalFrameDefault at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyFunction_Vectorcall at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
object_vacall at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyObject_CallMethodIdObjArgs at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
PyImport_ImportModuleLevelObject at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyEval_EvalFrameDefault at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
_PyEval_EvalCode at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
pyrun_file at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
pyrun_simple_file at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
PyRun_SimpleFileExFlags at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
pymain_run_file at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
pymain_run_python at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
Py_RunMain at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
pymain_main at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
main at /Users/Eric/opt/miniconda3/envs/general/bin/python (unknown line)
Allocations: 26161888 (Pool: 26156245; Big: 5643); GC: 30`

Is there anything that I missed?

Thanks for the help!

MMajidi137, 2021-08-04:
Hi @rockstaedt,
which version of Julia and Python do you use?

rockstaedt, 2021-08-04:
Julia 1.4 and Python 3.9

MMajidi137, 2021-08-04:
@rockstaedt which IDE do you use? Do you use Anaconda?

rockstaedt, 2021-08-04:
I used Pycharm and yes anaconda to create my environments :)

MMajidi137, 2021-08-04:
> I used Pycharm and yes anaconda to create my environments :)

 Pyjulia does not work well in Anaconda. My suggestion is to install Julia 1.5 and install Pycharm or spyder directly from pip (don't use anaconda)

rockstaedt, 2021-08-04:
Okay cool! Thanks! I'll try and let you know 👍🏼

SteffenMeinecke, 2021-11-11:
@rockstaedt does it work? Can this issue be closed?

rockstaedt, 2021-11-11:
To be honest, I did not try since I followed another approach.

closed 2021-11-11 by SteffenMeinecke