e2nIEE/pandapower #1312: Incorrect base power of MATPOWER test cases

ivanightingale, 2021-08-04:
Example:
```
import pandapower.networks as pn

net = pn.case5()
print(net.sn_mva)
```
prints `1`, but in the original case5,
```
mpc.baseMVA = 100;
```

This seems to result in incorrect per-unit impedance being calculated (such as in `pp.topology.create_nxgraph(net, calc_branch_impedances=True, branch_impedance_unit="pu")`).  In this case, manually setting `net.sn_mva = 100` solves the inconsistency.

SteffenMeinecke, 2021-08-06:
Several IEEE cases were converted when pandapower did not have the variable `net.sn_mva` or it was 1 by default. That grids still have the value `1` is a relic from that time. A first check showed that a change to `100 MVA` does not worsen the convergence properties. From my point of view, therefore, there is nothing to be said against correcting `net.sn_mva`. @ivanightingale can you check in which of the ieee_power_system_testcases which value for `baseMVA` is originally given? (There are also distribution grids which might have a different value.)

ivanightingale, 2021-08-06:
@SteffenMeinecke I checked the original MATPOWER cases corresponding to each case in pandapower/networks/power_system_test_case_jsons/.

`baseMVA = 10` (the file also contains `Sbase = mpc.baseMVA * 1e6`):
case33bw

`baseMVA = 100`:
case118, case1354pegase, case14, case145, case1888rte, case24_ieee_rts, case2848rte, case2869pegase, case30, case300, case3120sp, case39, case4gs, case5, case57, case6470rte, case6495rte, case6515rte, case6ww, case89pegase, case9, case9241pegase, case_ieee30

not in MATPOWER:
GBnetwork, GBreducednetwork, case11_iwamoto, case5_demo_gridcal, case_illinois200, iceland

SteffenMeinecke, 2021-08-10:
Thanks, @ivanightingale 
I checked the remaining grids:
- GBnetwork, GBreducednetwork, case_illinois200 and iceland have `baseMVA = 100`.
- for case11_iwamoto , which @WinfriedL added to pandapower, `baseMVA = 1` should be correct (in the paper powers of e.g. 0.128 pu are mentioned and in the pp net the net has p_mw=0.128).
- case5_demo_gridcal, which @jkrstulo added, comes from https://gridcal.readthedocs.io/en/latest/tutorials/five_node_grid.html GridCal has the baseMVA information not in the net but in the power flow function. There 100 MVA is the default. Due to that and the power values, I suppose to change also to `baseMVA = 100`.

commit c5408199 (2021-08-11): resave jsons of power_system_test_cases to correct net.sn_mva (baseMVA) close #1312
closed 2021-08-11 by SteffenMeinecke