e2nIEE/pandapower #1339: a (probably) silly question about loss calculation

shinespeciall, 2021-09-01:
on a 2 buses network i construct and run the test net like this:
```
net = pp.create_empty_network()
busnr1 = pp.create_bus(net, name="bus1", vn_kv=12.66, geodata=[0, 0])
busnr2 = pp.create_bus(net, name="bus2", vn_kv=12.66, geodata=[0, -1])
pp.create_ext_grid(net, busnr1)
pp.create_line_from_parameters(net, from_bus = busnr1, to_bus = busnr2,
                                length_km = 1, max_i_ka = 0.1966, r_ohm_per_km = 0.0922,
                                x_ohm_per_km = 0.047, c_nf_per_km = 0)
pp.create_load(net, busnr2, 0.100, 0.060)
pp.runpp(net)
```
then i use 2 ways to calculate the line loss but they give me different results:
```
print("pl_mw: ", sum(net.res_line.p_from_mw + net.res_line.p_to_mw) * 1000)
print("ql_mvar: ", sum(net.res_line.q_from_mvar + net.res_line.q_to_mvar) * 1000)
voltage_temp = net.res_bus.vm_pu[busnr1] - net.res_bus.vm_pu[busnr2]
theta_temp = math.radians(net.res_bus.va_degree[busnr1] - net.res_bus.va_degree[busnr2])
voltage_bus_temp = complex(voltage_temp * math.cos(theta_temp), voltage_temp * math.sin(theta_temp)) * 12.66 * 1000
loss_one_cur_line = voltage_bus_temp.conjugate() * voltage_bus_temp / complex(0.0922, 0.047).conjugate()
print(loss_one_cur_line / 1000)
```
output:
```
pl_mw:  0.007824699616618136
ql_mvar:  0.003988729739487495
(0.007787515299713699+0.003969774610483121j)
```
 
could anyone help and tell me what's the problem with my second method?

SteffenMeinecke, 2021-09-28:
you didn't consider the voltage angle.
Try this instead of your 2nd way:
```
vf = cmath.rect(net.res_bus.vm_pu.at[0], np.deg2rad(net.res_bus.va_degree.at[0]))
vt = cmath.rect(net.res_bus.vm_pu.at[1], np.deg2rad(net.res_bus.va_degree.at[1]))
vdrop = (vf - vt) * 12.66e3
Y = 1 / (0.0922 + 0.047j)
loss = Y * vdrop * vdrop.conjugate()
print(loss)
```

shinespeciall, 2021-09-29:
just try your code, you are right, thanks for help!

closed 2021-09-29 by SteffenMeinecke