e2nIEE/pandapower #1340: Inquiries related to the power flow results of DC lines

chochon, 2021-09-03:
Hello,
I am a new user of pandapower.
I am inquiring related with the power flow result of HVDC in my designed network. 

I modeled HVDC line in AC transmission network like this example network. 

[image]

Then, I simulated power flow. 

`pp.runpp(net, algorithm ='nr', init='auto', calculate_voltage_angles='auto', enforce_q_lims='True', max_iteration=20, trafo_loading="power")`

However, only the reactive power of generators are considered of their q_limit but dc lines are not as following:

<initial values>
[image]

<power flow results>
[image]

I hope dc lines are also operating within their reactive power limits. How can I solve this problem?
Please help me.

Thanks,

SteffenMeinecke, 2021-09-28:
Hi @chochon 
you should use `enforce_q_lims=True` instead of `enforce_q_lims='True'`. If it still doesn't work properly, you could state here the grid data for a complete minimal example.

SteffenMeinecke, 2021-11-11:
closed due to no further questions

closed 2021-11-11 by SteffenMeinecke