e2nIEE/pandapower #1349: Plot of bus index

hitchenzhe, 2021-09-22:
Hello,

My question is how to draw bus index without bus coordinates.
I created a new network, in which coordinates of bus is empty.
I need a plotting of network topo with bus indice. Any guidance? Thank you!

`lc = plot.create_line_collection(net, net.line.index, color=colors[1], zorder=0)  # create lines
bc = plot.create_bus_collection(net, net.bus.index, size=0.2, color=colors[0], zorder=1)  # create buses
sc = plot.create_bus_collection(net, net.ext_grid.bus.values, patch_type="rect", size=0.3, color=colors[2], zorder=11)
tc = plot.create_trafo_collection(net, net.trafo.index, color=colors[3], size=0.1)
plot.draw_collections([lc, bc, sc, tc], figsize=(3, 10))  # plot lines and buses`

SteffenMeinecke, 2021-09-28:
Hi @hitchenzhe 
you need to give pandapower information where to plot the buses if you want them to be plotted. You can do that on your own or you use the functionality provided by `pandapower.topology.generic_geodata.create_generic_coordinates()`. As an example you can have a look how this function is used in `pandpaower.plotting.simple_plot.simple_plot()`.

SteffenMeinecke, 2021-11-11:
closed due to no further questions

closed 2021-11-11 by SteffenMeinecke