e2nIEE/pandapower #1350: Problems with 3 phase asymmetric power flow

zkohth, 2021-09-24:
Hi all, I am trying to simulate a single phase load and perform a load flow calculation.
The load draws a power of 7.4 kW and 16 A with 230 V 
But the three phase load flow calculation returns empty tables and no results.
can anyone help ?
enclosed is my code.
thank you.

I always get this error message:

"/Users/zaidhasan/opt/anaconda3/lib/python3.8/site-packages/scipy/sparse/linalg/dsolve/linsolve.py:206: MatrixRankWarning: Matrix is exactly singular
  warn("Matrix is exactly singular", MatrixRankWarning)
/Users/zaidhasan/opt/anaconda3/lib/python3.8/site-packages/scipy/sparse/linalg/dsolve/linsolve.py:206: MatrixRankWarning: Matrix is exactly singular
  warn("Matrix is exactly singular", MatrixRankWarning)
/Users/zaidhasan/opt/anaconda3/lib/python3.8/site-packages/pandapower/pf/runpp_3ph.py:503: RuntimeWarning: invalid value encountered in greater
  while (s_mismatch > outer_tolerance_mva).any() and count < 30*max_iteration:"

----- CODE-----

import pandapower as pp
import math
from pandapower.pf.runpp_3ph import runpp_3ph

net=pp.create_empty_network()

bus_hv = pp.create_bus(net, vn_kv=20., name="Bus_hv ")
bus_lv = pp.create_bus(net, vn_kv=0.4, name="Bus_lv")
muff_1 = pp.create_bus(net, vn_kv=0.4, name="muff 1",type="m")

P= 0.0074
cos_phi=0.97
Phi=math.acos(cos_phi)
Q= (math.tan(Phi)*P)/1000000

#create bus elements
pp.create_ext_grid(net, bus=bus_hv, vm_pu=1.03, name="Grid Connection",s_sc_max_mva=1000,rx_max=0.1,x0x_max=0.1,r0x0_max=0.1)
pp. create_asymmetric_load (net , bus=muff_1 , p_a_mw = P ,q_a_mvar=Q)

L=0.012
L_HA=0.05
pp.create_transformer_from_parameters(net, hv_bus=bus_hv, lv_bus=bus_lv,name="Trafo_1", sn_mva= 0.4, vn_hv_kv=20, vn_lv_kv= 0.4, vk_percent=6, vkr_percent=1.425,pfe_kw=1.35,i0_percent=0.3375,shift_degree=150, vector_group='Dyn')
pp.create_line_from_parameters(net,from_bus =bus_lv, to_bus = muff_1, length_km=0.01, r_ohm_per_km = 0.642, x_ohm_per_km = 0.083, c_nf_per_km = 210, max_i_ka = 0.142,r0_ohm_per_km=0)

pp.add_zero_impedance_parameters(net)
pp.runpp_3ph(net)

elodin, 2021-11-01:
Hi @zkohth,
thanks for reporting your issue and providing the code.

I looked into it and can help to get it running. For three-phase calculation, several net elements need zero sequence parameters (ext_grid, trafo, line). You already provided some, but others are still missing. Because they are considered zero or NaN if they are not defined, the calculation runs into some problems. But the feedback you get from running the code is not giving you too much of a hint about what actually goes wrong, so it's good you started the issue. May I ask, what tutorial or documentation you used to create the code? This could help us improve the information and code-feedback in the future.

But now to the solution. I updated your code to make it work and attached it at the end of the comment. 
I put in the missing zero sequence parameters in the creation of the elements (trafo and line) and therefor can out comment 'add_zero_impedance_parameters(net)'. This function adds the zero seq. columns to the net to show you, what input parameters are necessary - but by adding them directly during the element-creation, the columns are automatically added anyway. Be aware though, that the values I used are just generic values to get the code to run, but don't represent any real world element values. Depending on how accurate your simulation should be, you may need to replace them with the correct values for your specific elements.

I hope this helps you. Please let me know if something remains unclear.

----- CODE -----
import pandapower as pp
import math
from pandapower.pf.runpp_3ph import runpp_3ph

net = pp.create_empty_network()
# pp.add_zero_impedance_parameters(net)

bus_hv = pp.create_bus(net, vn_kv=20., name="Bus_hv ")
bus_lv = pp.create_bus(net, vn_kv=0.4, name="Bus_lv")
muff_1 = pp.create_bus(net, vn_kv=0.4, name="muff 1", type="m")

P = 0.0074
cos_phi = 0.97
Phi = math.acos(cos_phi)
Q = (math.tan(Phi)*P) / 1000000

# create bus elements
pp.create_ext_grid(net,
                   bus=bus_hv,
                   vm_pu=1.03,
                   name="Grid Connection",
                   s_sc_max_mva=1000,
                   rx_max=0.1,
                   x0x_max=0.1,
                   r0x0_max=0.1)

pp. create_asymmetric_load(net,
                           bus=muff_1,
                           p_a_mw=P,
                           q_a_mvar=Q)

pp.create_transformer_from_parameters(net,
                                      hv_bus=bus_hv,
                                      lv_bus=bus_lv,
                                      name="Trafo_1",
                                      sn_mva=0.4,
                                      vn_hv_kv=20,
                                      vn_lv_kv=0.4,
                                      vk_percent=6,
                                      vkr_percent=1.425,
                                      pfe_kw=1.35,
                                      i0_percent=0.3375,
                                      shift_degree=150,
                                      vector_group='Dyn',
                                      vk0_percent=1,
                                      vkr0_percent=1,
                                      mag0_percent=100,
                                      mag0_rx=0,
                                      si0_hv_partial=0.9)

pp.create_line_from_parameters(net,
                               from_bus=bus_lv,
                               to_bus=muff_1,
                               length_km=0.01,
                               r_ohm_per_km=0.642,
                               x_ohm_per_km=0.083,
                               c_nf_per_km=210,
                               max_i_ka=0.142,
                               r0_ohm_per_km=1,
                               c0_nf_per_km=1,
                               x0_ohm_per_km=1)

runpp_3ph(net)

elodin, 2021-12-03:
@zkohth 
Does my previous answer solve your issue? If not, feel free to ask for any clarification. In case it does solve it, please close this issue. Thank you.

closed 2021-12-03 by zkohth